ttu/ruuvitag-sensor #5: Change which ble device to use via cli argument

mrwhale, 2017-04-11:
So firstly, nice work on this, made things very easy!

I've got this working on a CHIP but currently have 2 bluetooth devices (onboard + usb dongle) and device `hci0` is already being used for something else. Could we get a optional argument when using this to specifiy which hci device to use?

ttu, 2017-04-14:
Hi,

Made this change to [change-ble-device](https://github.com/ttu/ruuvitag-sensor/tree/change-ble-device) branch. Could you try that it works with your device? Added a general Config class that is passed to functions, so next time small changes like this are little bit easier to make. Just let me know if you need more info or have some suggestions how to improve this.

mrwhale, 2017-04-14:
👍 you the best, will test shortly

ttu, 2017-10-05:
Will be fixed with https://github.com/ttu/ruuvitag-sensor/issues/23

closed 2017-10-05 by ttu