ttu/ruuvitag-sensor #8: Calculation of acceleration is not correct

samuelphy, 2017-04-26:
According to the Data format 3 specification the acceleration value is a 2-complement int16 MSB first. One way to calculate the signed value is to use this:
```
# Example: twos_complment(0xFC18,16) = -1000
def twos_complement(value, bits):
    if (value & (1 << (bits - 1))) != 0: # Check if sign bit is set
        value = value - (1 << bits)      # Compute negative value
    return value
```

It might also be useful to get the data for each axle separately.

ttu, 2017-04-27:
Thanks for the example. Currently away from the computer, but will try to make this fix and add all acceleration values by the end of next week. After that could also merge data format 3 branch to master and make a new pypi release.

ttu, 2017-05-03:
Fixed calculations and added data for each axel (https://github.com/ttu/ruuvitag-sensor/commit/2a060850db3a48e0ce789a148e91a0ad9398d57d)

samuelphy, 2017-05-03:
Great!

closed 2017-05-03 by samuelphy