ttu/ruuvitag-sensor #10: Accuracy of the measurements has a bug when using python 2.7

Jussihoo, 2017-04-30:
when using python 2.7 the accuracy of the data what the _get_temperature, _get_humidity, _get_pressure etc. in the decoder.py is allways integered instead of float. 

One way to fix this problem in python2.7 is to add this as the first code line in the decoder.py:
from __future__ import division

ttu, 2017-05-03:
Thanks for the fix. (https://github.com/ttu/ruuvitag-sensor/commit/fed8b0ceea84e97d693af5b5c757ba407265b45d) Will merge this to master with other data format changes.

Should also change unit test code, so could run tests with Python 2.7

ttu, 2017-05-05:
Merged to master (https://github.com/ttu/ruuvitag-sensor/pull/12) and released with version 0.6.0

closed 2017-05-05 by ttu