ttu/ruuvitag-sensor #11: Unable to find tags and data

hemantkamalakar, 2017-05-05:
Hi,
Thanks for the ruuvitag-sensor library and good documentation for the Raspberry Pi.
I can see the RuuviTags (Flashed with latest weather station firmware) in nrfConnect and also in android nearby notifications. 
After successfully installing the ruuvitag-sensor library I tried running the find_tags.py which is not showing any tags. However get_ble_data.py shows the MAC id's of the ruuvitags.
What could be the issue?
print_to_screen.py results in follwing error in log file.

`2017-05-05 10:17:13,861 - ruuvitag_sensor - ERROR - Encoded value: BGgYAK447 not valid
Traceback (most recent call last):
  File "/root/.local/lib/python3.6/site-packages/ruuvitag_sensor/url_decoder.py", line 51, in decode_data
    decoded = bytearray(base64.b64decode(encoded, '-_'))
  File "/usr/local/lib/python3.6/base64.py", line 87, in b64decode
    return binascii.a2b_base64(s)
binascii.Error: Incorrect padding`

samuelphy, 2017-05-05:
I'm not sure here but I think you need to use the "data-format-3" branch of the repo instead to make it compatible with the latest RuuviTag FW.
Try installing with:
`pip3 install --upgrade git+https://github.com/ttu/ruuvitag-sensor@data-format-3`
I install it in my home folder so I add the `--user` argument as well.

hemantkamalakar, 2017-05-05:
Thanks a lot.
It worked.

closed 2017-05-05 by hemantkamalakar