ttu/ruuvitag-sensor #16: Error running Parse Example Code in Readme.md

jsavage, 2017-06-07:
The following is some example code under the heading of Parse Data within the Readme file:
```
from ruuvitag_sensor.ruuvi import RuuviTagSensor
from ruuvitag_sensor.decoder import UrlDecoder
full_data = '043E2A0201030157168974A5F41E0201060303AAFE1616AAFE10EE037275752E76692341412C3E672B49246AB9'
data = full_data[26:]
encoded = RuuviTagSensor.convert_data(data)
sensor_data = UrlDecoder().decode_data(encoded)
print(sensor_data)
```
When I run this code as follows , I get the following error:
```
$ python3 parse_data.py
Encoded value: (None, None) not valid
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.4/site-packages/ruuvitag_sensor/decoder.py", line 72, in decode_data
    decoded = bytearray(base64.b64decode(encoded, '-_'))
  File "/usr/lib/python3.4/base64.py", line 83, in b64decode
    s = _bytes_from_decode_data(s)
  File "/usr/lib/python3.4/base64.py", line 46, in _bytes_from_decode_data
    "string, not %r" % s.__class__.__name__) from None
TypeError: argument should be a bytes-like object or ASCII string, not 'tuple'
None
```

mikaelho, 2017-06-08:
Running Python 3.6.0, all I get with the above code is 'None'. No errors.

Which version of Python are you using?

ttu, 2017-06-08:
Forgot to update parse data example when added support for different data types. Also not sure if data in that example is valid. I'll have to inspect more.

https://github.com/ttu/ruuvitag-sensor/commit/b85047f68c475b296c51e1dd1eac0f3dedcc032e#diff-12c3cc89af19a6fa3f633f7fca29f169R57

Can you try if this works?
```python
from ruuvitag_sensor.ruuvi import RuuviTagSensor
from ruuvitag_sensor.decoder import UrlDecoder

full_data = '043E2A0201030157168974A5F41E0201060303AAFE1616AAFE10EE037275752E76692F23416A7759414D4663CD'
data = full_data[26:]

# convert_data returns a tuple which has data type id and encoded data
encoded = RuuviTagSensor.convert_data(data)

sensor_data = UrlDecoder().decode_data(encoded[1])

print(sensor_data)
```

jsavage, 2017-06-08:
Ok
That example seems to work although I suspect it may need example data with an identifier in it

pi@raspberrypi7:~ $ python3 parse_data2.py 
{'identifier': None, 'pressure': 995.0, 'humidity': 30.0, 'temperature': 24.0}

Thanks

ttu, 2017-06-08:
Old firmware sent base91 encoded data and that example is still from that time. Good that you noticed this, so we got correct example to README.

Updated example https://github.com/ttu/ruuvitag-sensor/commit/7fd32ecbbdbb0e40f6972743e754b40aa4f97eea

closed 2017-06-08 by ttu