ttu/ruuvitag-sensor #17: Fails to list or connect to Ruuvitags

rlaitner, 2017-08-04:
Hi,
I have a ruuvitag broadcasing about 10 feet from my pi; however, I can't connect to it. I get an error that says "set scan parameters failed: Input/Output error" when I run `python ruuvitag_sensor -f` from command line. Am I missing something obvious? It also appears in a third-party application known as Ruuvitag-logger that uses this as a dependency.

ttu, 2017-08-04:
Can you send me what you have in ruuvitag_sensor.log file? Here are few commands that we can use for troubleshooting.

What is the device when you type?
```sh
$ hcitool dev
```
Command should print something like this:
```
Devices:
        hci0    XX:XX:XX:XX:XX:XX
```
Does this command start to print MAC addresses?
```sh
$ hcitool lescan
```

rlaitner, 2017-08-05:
I didn't the log file originally. I figured it out. My pi had psutil installed for python 2.x, but not python 3.x. Thanks for your help. It works now.

closed 2017-08-05 by rlaitner

ttu, 2017-08-05:
Good that you got it working 👍 

Will try to add soon some troubleshooting documentation that will cover most common error cases.