ttu/ruuvitag-sensor #19: First code snippet on homepage fails, RuuviTag deprecated?

dcbb, 2017-08-18:
From the examples on the homepage, the first one seems to be based on a deprecated (or moved?) class:
   
    from ruuvitag_sensor.ruuvi import RuuviTag

fails with an import error.    

     ImportError: cannot import name 'RuuviTag'

`RuuviTagSensor` works fine.

ttu, 2017-08-18:
Thanks for noticing this! I made some small changes to classes during the last releast and I also updated the example code, but apparently didn't update everything.

I think the only update it needs is for the import line:
from ruuvitag_sensor.ruuvi import RuuviTag
->
from ruuvitag_sensor.ruuvitag import RuuviTag

```python
from ruuvitag_sensor.ruuvitag import RuuviTag

sensor = RuuviTag('AA:2C:6A:1E:59:3D')

# update state from the device
state = sensor.update()

# get latest state (does not get it from the device)
state = sensor.state

print(state)
```

ttu, 2017-08-18:
Fixed with https://github.com/ttu/ruuvitag-sensor/commit/a8dde9d41a88d385a058c87c7115b9e9fe1282c5

closed 2017-08-18 by ttu