ttu/ruuvitag-sensor #20: Negative values not decoded correctly from data format3

ssamppa, 2017-09-05:
Negative values seems to be incorrectly decoded. Last night was the first night with subzero temperatures at our home, see the picture. Crossing zero leads ruuvitag_sensor 0.8.0 to report -128.
Ideally add a test case for these boundary values :)

[image]

ttu, 2017-09-05:
Thanks for the info!

Unfortunately haven't been using much RuuviTag's data format 3, as my own home automation project hasn't been active in a while, so I haven't validated all different data cases.

Found a fix for the integer part of the temperature and still need to fix fractional part when I get more info of what the firmware will actually send (not sure if there is an error in decoding).

ssamppa, 2017-09-06:
Seems to be working fine now. Thanks for the quick fix!
[image]

ttu, 2017-09-06:
https://github.com/ttu/ruuvitag-sensor/commit/e6f2f40e535df29487ce5ccbc264e5c4d64fa505

closed 2017-09-06 by ttu