ttu/ruuvitag-sensor #23: Have a way to define the bluetooth device if more than one exists (linux)

kymwatts, 2017-10-03:
Hello,

I have more than one bluetooth device on my linux machine.
BleCommunicationNix.start() will hang, as hci0 is not a usable bluetool device.

Ideally, id like to use hci1, which is my main bluetooth dongle.
Is it possible to have an environment var to set alternatives or a argument when instancing RuuviTagSensor?

Cheers
Kym

ttu, 2017-10-03:
Hi.

I had this change almost completed, but last time didn't get anyone to test it. Made some minor changes and added it to the new branch: https://github.com/ttu/ruuvitag-sensor/tree/define-bt-device

Could you test it from there?

kymwatts, 2017-10-14:
I have confirmed this works.
Thanks

ttu, 2017-10-15:
Thanks for testing.

Merged to master https://github.com/ttu/ruuvitag-sensor/commit/4c981f0b32ee0348ca85dcb7518c12d3df9d561d

closed 2017-10-15 by ttu