ttu/ruuvitag-sensor #26: cannot find psutil on exit(?)  -( installed as user xxx not pi)

DG12, 2017-10-16:
>
-l
Exception ImportError: 'No module named psutil' in <generator object get_datas at 0x76b72030> ignored

-f too (shows traceback ending with:
File "/usr/local/lib/python2.7/dist-packages/ruuvitag_sensor/ble_communication.py", line 66, in stop
    import psutil
ImportError: No module named psutil

>
ls -l /usr/local/lib/python2.7/site-packages/
total 0

**Should I just move /home/dgerman/.local/lib/python3.4/site-packages to /usr/local...??**

>
piw:/home/dgerman > ls -l .local/lib/python3.4/site-packages/. #notice NOT root .local
total 28
drwxr-x---  2 dgerman staff 4096 Oct 15 12:06 Rx-1.6.0.dist-info
drwxr-x---  4 dgerman staff 4096 Oct 15 18:09 psutil
drwxr-x---  2 dgerman staff 4096 Oct 15 12:06 psutil-5.4.0.egg-info
drwxr-x---  3 dgerman staff 4096 Oct 15 19:44 ruuvitag_sensor
drwxr-x---  2 dgerman staff 4096 Oct 15 18:25 ruuvitag_sensor-0.8.2.egg-info
-rw-r-----  1 dgerman staff   78 Oct 15 19:48 ruuvitag_sensor.log
drwxr-x--- 12 dgerman staff 4096 Oct 15 12:06 rx

>
piw:/usr/bin > uname -a
Linux piw 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux
>
piw:/usr/bin > cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

============================================
**By the way, are these links correct in particular**  python -> python2.7
> lrwxrwxrwx 1 root root       8 Mar 28  2015 pydoc -> pydoc2.7
-rwxr-xr-x 1 root root      79 Sep 17  2016 pydoc2.7
lrwxrwxrwx 1 root root       8 Dec  6  2014 pydoc3 -> pydoc3.4
-rwxr-xr-x 1 root root      79 Oct 19  2014 pydoc3.4
lrwxrwxrwx 1 root root      12 Mar 28  2015 pygettext -> pygettext2.7
-rwxr-xr-x 1 root root   22097 Sep 17  2016 pygettext2.7
lrwxrwxrwx 1 root root      12 Dec  6  2014 pygettext3 -> pygettext3.4
-rwxr-xr-x 1 root root   22368 Oct 19  2014 pygettext3.4
lrwxrwxrwx 1 root root      22 Mar  7  2016 pypy -> ../lib/pypy/bin/pypy-c
-rwxr-xr-x 1 root root    4076 Nov 20  2015 pypyclean
-rwxr-xr-x 1 root root    3511 Nov 20  2015 pypycompile
lrwxrwxrwx 1 root root       9 Mar 28  2015 python -> python2.7
lrwxrwxrwx 1 root root       9 Mar 28  2015 python2 -> python2.7
-rwxr-xr-x 1 root root 3201580 Sep 17  2016 python2.7
lrwxrwxrwx 1 root root       9 Dec  6  2014 python3 -> python3.4
lrwxrwxrwx 1 root root      16 Dec  6  2014 python3-config -> python3.4-config
-rwxr-xr-x 2 root root 3816928 Oct 19  2014 python3.4
lrwxrwxrwx 1 root root      36 Oct 19  2014 python3.4-config -> arm-linux-gnueabihf-python3.4-config
-rwxr-xr-x 2 root root 3816928 Oct 19  2014 python3.4m
lrwxrwxrwx 1 root root      37 Oct 19  2014 python3.4m-config -> arm-linux-gnueabihf-python3.4m-config
lrwxrwxrwx 1 root root      10 Dec  6  2014 python3m -> python3.4m
lrwxrwxrwx 1 root root      17 Dec  6  2014 python3m-config -> python3.4m-config
lrwxrwxrwx 1 root root      29 Mar 28  2015 pyversions -> ../share/python/pyversions.py

> 
Thanks for your help

ttu, 2017-10-16:
What command did you use to install packages? (pip or pip3)

Error has a python2.7 in the path but installed packages are under python3.4
```
File "/usr/local/lib/python2.7/dist-packages/ruuvitag_sensor/ble_communication.py", line 66, in stop
```

DG12, 2017-10-16:
pip3 install --user ruuvitag-sensor

as per
https://github.com/ttu/ruuvitag-sensor/blob/master/install_guide_pi.md#install-ruuvitag-sensor-package

Thanks for the reply

ttu, 2017-10-17:
Did you run it with Python2 or Python3?

How about if you try to make a virtualenv with a defined python version and execute the verification script, e.g. in https://github.com/ttu/ruuvitag-sensor/blob/master/verification.sh#L17

ttu, 2018-01-07:
@DG12 are you still having this problem?

robertsicoie, 2018-02-05:
@DG12 If you run it with python3, you should install python3-psutil:`sudo apt-get install python3-psutil`
I faced the same problem and this fixed it.

ttu, 2018-02-07:
@robertsicoie thanks for the info! Do you think that we should change from the README psutil installation intruction to include also python-psutil and python3-psutil?

* Python 2.7 and 3
    * psutil
        * Package uses psutil to start and stop processes. Psutil requires `sudo apt-get install python-dev python-psutil` or `sudo apt-get install python3-dev python3-psutil`

robertsicoie, 2018-02-07:
Yes, definitely.

ttu, 2018-02-09:
Closed with commit ae28d61f1ce5671816bbee67084388c3c4768caf

closed 2018-02-09 by ttu