ttu/ruuvitag-sensor #31: hcitool deprecated upstream

ritman, 2017-11-24:
It seems upstream does not include ``hcitool`` anymore in bluez-utils [1]. Sooner or later people will realize this when they run into problems. Although there are legacy packages already for some distros (although not official), I think it might be a good idea to add support for an alternative method to communicate using bluetooth. ``bluetoothctl`` is the upstream alternative and therefore might be a good candidate.

[1] https://www.spinics.net/lists/linux-bluetooth/msg69239.html

ttu, 2017-11-26:
Thanks for the info, will check bluetoothctl.

An option that I was checking earlier was to switch the Bluetooth communication from hcitool to python socket module. This way the package would not be dependant on any external libraries.  I haven't found any good examples yet, so I don't have any idea how easy it is to implement or will it even work.

ukBaz, 2017-11-26:
You might want to take a look at the great work @frawau is doing. 
https://github.com/frawau/aioblescan/tree/master/aioblescan

ttu, 2017-11-26:
Good to know that it will work :) Will check basic configuration from there. Good work @frawau!

ttu, 2017-12-06:
Unfortunately can't use socket module as by default Python does not support socket.AF_BLUETOOTH. 

E.g. with rasbpian you will get this error
> module 'socket' has no attribute 'AF_BLUETOOTH'

https://docs.python.org/3/library/socket.html#socket-families
> Depending on the system and the build options, various socket families are supported by this module.

Apparently Python needs to be compiled with bluez headers to make it work.

Will check `bluetoothctl` or some cross platform library e.g. https://github.com/TheCellule/python-bleson.

ukBaz, 2017-12-06:
What version of Python are you using? I have not done anything special on the latest version of Raspbian and it works with Python3.

ttu, 2017-12-06:
Ok, good to know that the problem was with my own environment. I used Python 3.6.0. I have an old Raspbian where I installed it manually, so that is likely then the problem. Will re-install newer Raspbian.

I will continue checking [Bleson](https://github.com/TheCellule/python-bleson), as it is cross-platform and will provide also macOS and Windows support. Unfortunately it is still in early alpha, but @WayneKeenan seems to be actively developing it. It also uses socket module on Linux.

cmarxer, 2018-01-24:
I guess, `hciconfig` is also affected.

It's used in`ble_communication.py`, line 67:
`subprocess.call('sudo hciconfig %s reset' % bt_device, shell=True, stdout=DEVNULL)`

According to the arch linux wiki, `btmgmt` is the new tool to use:
[https://wiki.archlinux.org/index.php/bluetooth#Shell_command_is_missing_from_bluez-utils](https://wiki.archlinux.org/index.php/bluetooth#Shell_command_is_missing_from_bluez-utils)

DG12, 2018-02-27:
Although raspberry pi stretch release does not include hcitool it can be retrieved using:
` sudo apt-get install bluetooth bluez blueman bluez-hcidump`

Landrash, 2018-03-21:
I would recommend having a look at https://github.com/open-homeautomation/miflora that's for another bluetooth LE device :)

ttu, 2018-04-18:
Added first implementation with Bleson to own branch: https://github.com/ttu/ruuvitag-sensor/tree/bleson-ble-communication

Works with Linux. On Windows finds ble devices, but doesn't return any data. Haven't tried with macOS.

Must be installed with dependency-links as Bleson is installed from GitHub
```sh
$ pip install -e . --process-dependency-links
```

tazle, 2018-04-21:
It seems that Bleson only provides one BLE advertisement packet per tag. Duplicates are disabled by default on the Linux adapter, and there doesn't seem to be any way to enable them in the API.

tazle, 2018-04-21:
Raised and created PR

ttu, 2020-04-09:
Optional Bleason communicaton released. Check [README](https://github.com/ttu/ruuvitag-sensor#bleson) and [issue #78](https://github.com/ttu/ruuvitag-sensor/issues/78).

closed 2020-04-09 by ttu