ttu/ruuvitag-sensor #32: hcitool and hcidump is not terminated

samuelphy, 2017-11-26:
I’ve noticed that when my Python code is executed from rc.local (i.e. at startup) the RuuviTagSensor.get_data_for_sensors() is firing off new processes that are not terminated. I.e. each time RuuviTagSensor.get_data_for_sensors() is called `hcitool lescan` and `hcidump` are executed again and again. This memory leak fills up the RAM… I.e. I get multiple versions of the processes below:
```
pi@raspberry:~ $ ps 2188
PID TTY STAT TIME COMMAND
2188 ? S 0:00 hcitool lescan --duplicates
pi@raspberry:~ $ ps 2231
PID TTY STAT TIME COMMAND
2231 ? S 0:00 sudo -n hcidump --raw
```
Note: This does not happen when I execute the Python code on the command line using nohup, doesn’t matter if the user is pi or root. I guess the error lies in:
`def stop(hcitool, hcidump):`
in ble_communication.py but I have not looked further for now.

samuelphy, 2017-11-27:
I think I have missed installing `psutil`. I'll verify and close if it fixes the issue.

samuelphy, 2017-11-27:
psutil was already installed but the error was on my part having the "&" operand at the end of the command line in rc.local.

closed 2017-11-27 by samuelphy

aapris, 2017-12-29:
I see the same when I run my script from crontab's @reboot. Finally `hci0` hangs and I get no readings from RuuviTags. This happened twice yesterday on two different Raspi.

This may or may not help (the other Raspi needed reboot):
```
sudo hciconfig hci0 reset
sudo hciconfig hci0 up
```
I've 4 seconds timeout in this command:
`RuuviTagSensor.get_data_for_sensors([], timeout_in_sec)`
Maybe it should be longer?

Here is part of my `ps aux | grep hci`, there were tens of similar rows.

```
root     23519  0.0  0.3   7632  3472 ?        S    05:36   0:00 sudo -n hcitool lescan --duplicates
root     23525  0.0  0.0   2480   668 ?        S    05:36   0:12 hcitool lescan --duplicates
root     23546  0.0  0.3   7632  3412 ?        S    Dec24   0:00 sudo -n hcitool lescan --duplicates
root     23553  0.0  0.0   2480   632 ?        S    Dec24   2:11 hcitool lescan --duplicates
root     25215  0.0  0.3   7632  3360 ?        S    Dec22   0:00 sudo -n hcitool lescan --duplicates
root     25220  0.0  0.0   2480   640 ?        S    Dec22   2:35 hcitool lescan --duplicates
root     25452  0.0  0.3   7632  3356 ?        S    Dec27   0:00 sudo -n hcitool lescan --duplicates
root     25460  0.0  0.0   2480   660 ?        S    Dec27   1:17 hcitool lescan --duplicates
```

jareware, 2018-12-06:
I'm not detaching with `&` in my `rc.local` but I'm still getting the runaway `hcitool` processes:

[image]

(the drop is a reboot)

```
$ ps aux | grep hcitool | wc -l
149
```

I did notice though that I'm running 0.9.0, not the latest 0.11.0, will update and see if it has any effect.

aapris, 2018-12-06:
Newest version supports newest Ruuvitag data formats, so if you upgrade firmware, you need not upgrade ruuvitag_sensor also. I've this in root's crontab:
```
58 * * * *	killall hcitool; hciconfig hci0 reset && hciconfig hci0 up
0 * * * *	/usr/bin/supervisorctl restart RuuviTag
```
It kills all runaway hcitool processes and then (after 2 minutes) restarts my reader process.

jareware, 2018-12-07:
Yeah, I've "solved" this in a similar manner, with a scheduled reboot once per week.

jareware, 2018-12-08:
[image]

No effect.

aapris, 2018-12-08:
Hmm, it seems after upgrading to 0.11.0 my RuuviTag data has a lot of gaps at two separate locations. This happened weeks ago after I upgraded `ruuvitag-sensor` library. This can also be an issue in my hardware (Raspberry pi 3) so I've to investigate more of this to be sure what is the root cause.
[image]

aapris, 2018-12-08:
Could this be a problem in Raspbian or hcitool? Maybe some upgrade messed this up?

oduo, 2019-01-13:
I have also this issue. I use ruuvitag-sensor only from ruuvi-hass https://github.com/peltsippi/ruuvi-hass

When running long time I have multiple hcitool lescan --duplicates running on rassberry pi

root     26326  0.0  0.0   2480   624 ?        S    Jan09   0:32 hcitool lescan --duplicates
root     27096  0.0  0.3   7228  3324 ?        S    Jan11   0:00 sudo -n hcitool lescan --duplicates
root     27103  0.0  0.0   2480   668 ?        S    Jan11   0:15 hcitool lescan --duplicates
root     27217  0.0  0.3   7228  3368 ?        S    Jan06   0:00 sudo -n hcitool lescan --duplicates
root     27224  0.0  0.0   2480   668 ?        S    Jan06   0:46 hcitool lescan --duplicates
root     27456  0.0  0.3   7228  3260 ?        S    Jan09   0:00 sudo -n hcitool lescan --duplicates
root     27463  0.0  0.0   2480   668 ?        S    Jan09   0:30 hcitool lescan --duplicates
root     27626  0.0  0.3   7228  3256 ?        S    Jan08   0:00 sudo -n hcitool lescan --duplicates

pip list
ruuvitag-sensor         0.11.0
psutil                  5.4.8

(homeassistant) homeassistant@hassbian:~ $ uptime
 15:59:39 up 28 days, 34 min,  1 user,  load average: 0.02, 0.31, 0.30
(homeassistant) homeassistant@hassbian:~ $ pgrep hcitool|wc -l
116

Something wrong with process killing?

sevesalm, 2021-03-30:
I was able to solve my similar issues by calling the `stop()` method for the `RuuviTagReactive` object. Please check issue #125 for details.