ttu/ruuvitag-sensor #33: Add helper functions

ttu, 2017-12-17:
* Is Tag present
  * Static RuuviTagSensor function
  * Method for RuuviTag class

ghsriram, 2018-09-23:
Hello. I am using this example line but I still find my code hanging endlessly if the ruuvitag BLE is out of reach from my Pi. Any idea? I simply using stock code from ruuvitag-logger.py. Python version 3.6.
Thanks in advance. 

datas = RuuviTagSensor.get_data_for_sensors(macs, timeout_in_sec)

ttu, 2018-09-25:
Hi @ghsriram! Can you create a new issue from this BLE error?

Won't make any fixes for BLE communication for this old implementation, but you can try the implemenation with Bleson BLE communication. Check more info from [README](https://github.com/ttu/ruuvitag-sensor#requirements).

ghsriram, 2018-09-28:
Thanks much. I will try that.

closed 2020-01-25 by ttu