ttu/ruuvitag-sensor #35: binascii.Error occasionally

aapris, 2018-01-04:
I use RuuviTagSensor in my own script and it seems to crash every now and then in call 
`datas = RuuviTagSensor.get_data_for_sensors(macs, timeout_in_sec)`
(macs is [], timeout is 4) with traceback below:

```
Traceback (most recent call last):
  File "/home/pi/.virtualenvs/ruuvi/lib/python3.5/site-packages/ruuvitag_sensor/decoder.py", line 72, in decode_data
    decoded = bytearray(base64.b64decode(encoded, '-_'))
  File "/home/pi/.virtualenvs/ruuvi/lib/python3.5/base64.py", line 88, in b64decode
    return binascii.a2b_base64(s)
binascii.Error: Incorrect padding
Encoded value: w not valid
```
I'm trying to avoid this (from now) by catching this exception in my code, but I think this should be handled in RuuviTagSensor library, which could raise a custom exception or something.

What do you think?

ttu, 2018-01-04:
This is already handled in the library code, so it might be that error is just written to stdout, but the execution continues.
https://github.com/ttu/ruuvitag-sensor/blob/master/ruuvitag_sensor/decoder.py#L79

Is there anything else in the log file?

I wonder why RuuviTag broadcasts https://ruu.vi/#w? Maybe there is something wrong with the conversion of hex data.

aapris, 2018-01-04:
Ok, I'll update when the new version appears in pypi.

No, just those tracebacks.

Perhaps those are just interference in radio wavers or something?

aapris, 2018-01-08:
I forgot to mention: all RuuviTags are in high precision mode so they are not broadcasting URL.

ttu, 2018-01-14:
Then it seems that there is something wrong with the data conversion.
https://github.com/ttu/ruuvitag-sensor/blob/master/ruuvitag_sensor/ruuvi.py#L169

Could you add a print statement before this line and keep it running until it crashes? 
https://github.com/ttu/ruuvitag-sensor/blob/master/ruuvitag_sensor/ruuvi.py#L162

With that I could try to check what is wrong with the conversion function.

aapris, 2018-01-15:
I got this after running the script several hours:
`Z^Y*I`^X`
I guess `^Y` and `^X` are control characters.

ttu, 2018-01-17:
Sorry, my bad, I gave a bad guidance. 

Could you add this line `print(ble_data[1])` to  line 162. I would like to see original data so I could maybe fix `convert_data` function.

rwfnf, 2018-05-16:
I was getting the same issue, here is the result from print(ble_data[1]) on line 160:

```
May 16 21:50:25 abb python3[30625]: {'acceleration_y': 773, 'acceleration_z': 27, 'acceleration': 998.8403275799391, 'acceleration_x': -632, 'temperature': 21.93, 'humidity': 40.0, 'pressure': 1028.4, 'battery': 3091}2018-05-16 21:50:25,567  ERROR -               decoder.py: 94 -              decode_data(): Encoded value: \xcd\xe8 not valid
May 16 21:50:25 abb python3[30625]: Traceback (most recent call last):
May 16 21:50:25 abb python3[30625]:   File "/home/daa/ruuvi/ruuvitag-sensor/ruuvitag_sensor/decoder.py", line 86, in decode_data
May 16 21:50:25 abb python3[30625]:     decoded = bytearray(base64.b64decode(encoded, '-_'))
May 16 21:50:25 abb python3[30625]:   File "/usr/lib/python3.5/base64.py", line 88, in b64decode
May 16 21:50:25 abb python3[30625]:     return binascii.a2b_base64(s)
May 16 21:50:25 abb python3[30625]: binascii.Error: Incorrect padding
May 16 21:50:30 abb python3[30625]: 2018-05-16 21:50:30,567  ERROR -               decoder.py: 94 -              decode_data(): Encoded value: \xcd\xe8 not valabb
May 16 21:50:30 abb python3[30625]: Traceback (most recent call last):
May 16 21:50:30 abb python3[30625]:   File "/home/daa/ruuvi/ruuvitag-sensor/ruuvitag_sensor/decoder.py", line 86, in decode_data
May 16 21:50:30 abb python3[30625]:     decoded = bytearray(base64.b64decode(encoded, '-_'))
May 16 21:50:30 abb python3[30625]:   File "/usr/lib/python3.5/base64.py", line 88, in b64decode
May 16 21:50:30 abb python3[30625]:     return binascii.a2b_base64(s)
May 16 21:50:30 abb python3[30625]: binascii.Error: Incorrect padding
May 16 21:50:35 abb python3[30625]: 2018-05-16 21:50:35,573  ERROR -               decoder.py: 94 -              decode_data(): Encoded value: \xcd\xe7 not valabb
May 16 21:50:35 abb python3[30625]: Traceback (most recent call last):
May 16 21:50:35 abb python3[30625]:   File "/home/daa/ruuvi/ruuvitag-sensor/ruuvitag_sensor/decoder.py", line 86, in decode_data
May 16 21:50:35 abb python3[30625]:     decoded = bytearray(base64.b64decode(encoded, '-_'))
May 16 21:50:35 abb python3[30625]:   File "/usr/lib/python3.5/base64.py", line 88, in b64decode
May 16 21:50:35 abb python3[30625]:     return binascii.a2b_base64(s)
May 16 21:50:35 abb python3[30625]: binascii.Error: Incorrect padding
```

ttu, 2019-05-31:
Can't reproduce this error anymore. Decoder and payload handling has been updated after this issue was opened. If this error appears again, let's open a new issue.

closed 2019-05-31 by ttu