ttu/ruuvitag-sensor #37: Stupid Raspberry Zero W and wireless on Raspbian

hacklover, 2018-02-18:
Followed the guide and everything was working, after reboot my Raspberry Zero W wireless wont reconnect. Raspberry Zero W is a fucking waste of patience, why the fuck everything is ok and after reboot it can't connect to wifi anymore? Fourth time.

samuelphy, 2018-02-18:
This does not seem to be related to ruuvitag-sensor at all, but since I'm a nice guy I can give you a tip. Are you having trouble with the Avahi daemon at startup? If so, just download the latest Raspian version and it will fix your problem.

hacklover, 2018-02-19:
I know that and I'm sorry for the rage, feel free to delete this issue after all... ty for the tip, anyway I'm unable to download it because first I have to fix the static ip / dhcp trouble, but I don't know how. On web I found also other guys having problems with Stretch on Zero W :/

Anyway I can make sure you that the guide and code, before rebooting for the first time, is working good on Raspberry Zero W and isn't your fault

ttu, 2018-02-22:
@hacklover Hope you got your problems fixed. It is known problem with all IT-systems that these things just don't work, ever. 

Knowing that, raging is pretty understandable. Although Github issue is not the best place for that, no problems.

closed 2018-02-22 by ttu