ttu/ruuvitag-sensor #39: Cannot change bluetooth device

peltsippi, 2018-03-30:
Raspberry pi 3 B has a problem with built in bluetooth.

hciconf

> hci1:   Type: BR/EDR  Bus: UART
>         BD Address: B8:27:EB:F3:E7:01  ACL MTU: 1021:8  SCO MTU: 64:1
>         UP RUNNING PSCAN
>         RX bytes:9724 acl:0 sco:0 events:511 errors:0
>         TX bytes:7207 acl:0 sco:0 commands:432 errors:0

My test scrip (copypaste from examples, tried both hci1 mac and ruuvitag mac, same outcome)

> from ruuvitag_sensor.ruuvi import RuuviTagSensor
> from ruuvitag_sensor.ruuvitag import RuuviTag
> 
> sensor = RuuviTag('B8:27:EB:F3:E7:01', 'hci1')
> 
> RuuviTagSensor.find_ruuvitags('hci1')
> 
> datas = RuuviTagSensor.get_data_for_sensors(bt_device='hci1')
> 
> RuuviTagSensor.get_datas(lambda x: print('%s - %s' % (x[0], x[1]), bt_device=device))
> 

Output when test script is executed:

> Can't get device info: No such device
> Can't get device info: No such device
> 

For example hcidump seems to work just fine:

>
> pi@raspberrypi:~/ruuvitag $ hcidump -i hci1
> HCI sniffer - Bluetooth packet analyzer ver 5.23
> device: hci1 snap_len: 1500 filter: 0xffffffff
>
>

peltsippi, 2018-03-30:
Here is also output from ruuvitag-sensor when trying to change device through flag

> 
>  python3.6 ~/.local/lib/python3.6/site-packages/ruuvitag_sensor -f -d hci1
> Finding RuuviTags. Stop with Ctrl+C.
> Start receiving broadcasts (device hci0)
> Can't get device info: No such device
> End Of File (EOF). Exception style platform.
> Stop receiving broadcasts
>

ttu, 2018-03-30:
Thanks for the logs. I can see that it is not setting the device correctly and tries to use the default device. 

> Start receiving broadcasts (device hci0)

You can try fixed version from branch https://github.com/ttu/ruuvitag-sensor/tree/fix-btdevice-parameter

ttu, 2018-04-19:
Merged the branch with the fix, so can add it to the next release. If you still get the same error, please reopen the issue.

closed 2018-04-19 by ttu