ttu/ruuvitag-sensor #40: Bug: incompatible with RuuviFW 1.2.8

ojousima, 2018-04-14:
The 1.2.8 update to Ruuvi Firmware trims extra NULLs at the end of transmission which breaks the data format type check. I can fix this and implement #29 .

ttu, 2018-04-15:
Ok, that is great! I will label #29 as in progress.

ersiko, 2018-04-18:
Hi
Trying to help with my experience.
I'm using ruuvitag-sensor v0.10.0 and after upgrading my Ruuvi Tag to RuuviFW 1.2.8, ruuvitag-sensor stopped working. The update function doesn't show any error, just returns an empty dict. 
I read the new FW changed the default URL mode to a RAW mode, more precise and less battery-consuming (I get two decimals with my cell phone). I moved back to URL mode (pressing B button once) and ruuvitag-sensor library worked again (but less precise, no decimals).

ttu, 2018-04-19:
@ersiko Noticed that yesterday as I implemented BLE communication with Bleson. Although first though that it is just some difference how Bleson and hcidump processed the data. 

You can try this version if you want to get RuuviFW 1.2.8 RAW mode to work. https://github.com/ttu/ruuvitag-sensor/issues/31#issuecomment-382550336

closed 2018-04-19 by ttu