ttu/ruuvitag-sensor #47: acceleration, how can I get it?

bryantlawmy, 2018-06-18:
Possible with the package, I can get the acceleration value?

ttu, 2018-06-18:
Depends on RuuviTags firmware and which mode is activated (RAW mode sends also acceleration data).

https://github.com/ttu/ruuvitag-sensor#data-formats

https://lab.ruuvi.com/ruuvitag-fw/

bryantlawmy, 2018-06-18:
Cool, currently I am using one of your example link https://github.com/ttu/ruuvitag-sensor/blob/master/examples/print_to_screen.py where it only return as above but not acceleration, show I define the data formats?

ttu, 2018-06-18:
Press Button B from your RuuviTag and red light should start blinking dimly. It means that RAW mode is activated and then it will broadcast data in format 3 or 5 depending on the firmware. (https://lab.ruuvi.com/ruuvitag-fw/)

Then returned data dictionary will have `data['acceleration']`. If you want that example will also print it, you need to add that line manually.

bryantlawmy, 2018-06-19:
Excellent.

closed 2018-06-19 by bryantlawmy