ttu/ruuvitag-sensor #48: Running 2 python scripts for differrent ruuvitag sensor hit error

bryantlawmy, 2018-07-12:
As I running single python scripts work perfectly with single sensor, apparently, I using duplicate the same scripts as "print2.py" after for couple minutes, I hit below error for both python scripts:

Can't init device hci0: Device or resource busy (16)

bryantlawmy, 2018-07-12:
attached with my python script
[print2.py.txt](https://github.com/ttu/ruuvitag-sensor/files/2188908/print2.py.txt)

ttu, 2018-07-14:
Thanks for the example script.

We should add retry for bluetooth connection in case of failures like this.

Can you try if you can repeat the error with the code from [Bleson branch](https://github.com/ttu/ruuvitag-sensor/tree/bleson-ble-communication)?

ttu, 2019-06-14:
Unfortunately this is more like a feature with the Bluez implementation at the moment.

Before starting scanning processes Bluetooth device is reseted and if there is 2 different processes starting a new scanning at the same time it will eventually end up with an error.

When fetching data for multiple sensors it is recommended to use get_datas-method and provide all macs in an array.

closed 2019-06-14 by ttu