ttu/ruuvitag-sensor #52: RSSI

hongell, 2018-08-14:
Hi,

can you add RSSI to the list ?

A quick hack lto Df3Decoder like this seems to work:

    def _get_rssi(self, data):
        return data[-1]-256

                'rssi': self._get_rssi(byte_data)

{'acceleration': 1038.0934447341433, 'pressure': 1008.27, 'temperature': 24.75, 'acceleration_y': -13, 'acceleration_x': 5, 'battery': 3061, 'acceleration_z': 1038, **'rssi': -68,** 'humidity': 42.5}

Scrin, 2018-10-05:
You can parse the report length from the data and see if there's one "extra byte" at the end (=RSSI), from my notes (a data format 3 packet in this case):

```format 3
> 04 3E 25 02 01 03 01 D5 07 82 61 3C F2 19 02 01 04 15 FF 99 04 03 2C 1A 08 C9 79 00 0B FF F5 03 EB 0A ED 00 00 00 00 CD
# PT EC LE SE NR ET PA MAC-------------- RL SL ST FL SL ST COMID DF DATA---------------------------------------------- RS

PT = Packet type (04 = event)
EC = Event code (3E = LE Meta)
LE = Packet length
SE = Sub event (02 = LE Advertising Report)
NR = Number of Reports
ET = Event Type (03 = Non-Connectable undirected advertising)
PA = Peer Address Type (01 = Random Device Address)
RL = (next) Report length

SL = AD Struct Length
ST = AD Struct Type (01 = Flags, 03 = Service UUID, FF = Manufacturer specific data)
FL = AD Struct for flags
COMID = Company Identifier (99 04 is 0x0499 = Ruuvi Innovations Ltd.)

DF = Data Format
DATA = The actual "measurement data" from the tag

RS = RSSI
```

Most importantly:
- Third byte of the entire data (LE in my notes) is the **remaining** length of the packet from hcidump, **including the RSSI if it's present.**
- The fifth byte (NR) is the number of reports, for the normal broadcast data from ruuvitags this is always (?) 1, at least with the current firmwares
- The first byte of a report is the **remaining** length of **that report** (RL in my notes), the first report of a packet starts on the 14th byte, right after the MAC address.

So the most robust solution: parse the number of reports, then parse the length of all reports, and see if there's one extra byte after the last report (or compare it to the Packet length (LE in my notes) as the LE includes the RSSI byte if present)

The quickest solution is to check the 14th byte (remaining length of the first report) and see if the remaining length of the data is the same or 1 more (this will work only if there's one report per packet)

RodrigoD27avila, 2020-10-26:
Hi, my team and I are working on a smart basket and we're using your library.

For protocols 3 and 5 it was possible to extract the RSSI value in the last byte of the packet, and in our tests this proved to be useful to measure the distance later.

This is a quick solution, as we only take the last byte.

terop, 2022-08-07:
@ttu I think this issue can now be closed as the RSSI reporting is merged.

ttu, 2022-08-07:
Yes. Implementation in PR https://github.com/ttu/ruuvitag-sensor/pull/161

closed 2022-08-07 by ttu