ttu/ruuvitag-sensor #53: Add timezone support

latenssi, 2018-09-25:
At the moment `time` field is set to mirror the current local time of the recipient device without any information of the local timezone.

Either add timezone information or convert `time` to UTC.

I will try to make a PR but am currently really limited on time (who isn't) to contribute so don't hold your breath.

ttu, 2018-10-02:
@latenssi can you specify which `time` field you mean? The one in [ruuvi_rx.py](
https://github.com/ttu/ruuvitag-sensor/blob/287cc3ba28a589156ced180b9f74e1bb2ad6f007/ruuvitag_sensor/ruuvi_rx.py#L22)?

latenssi, 2018-10-02:
Yes that is the one I believe. Currently in my own application I am replacing the value given by [ruuvi_rx.py](https://github.com/ttu/ruuvitag-sensor/blob/287cc3ba28a589156ced180b9f74e1bb2ad6f007/ruuvitag_sensor/ruuvi_rx.py#L22) with `data[1]["time"] = datetime.datetime.utcnow().isoformat()` and this fixes the issue for me.

latenssi, 2018-10-02:
I made a proposal PR. I am not sure which is better; add timezone info or just use UTC as my PR does. Atleast what comes to simplicity I think my solution is better since no parsing of the devices timezone is required.