ttu/ruuvitag-sensor #57: Get sensor datas with callback hangs with the newest firmware version

fommike, 2018-10-17:
Hey, 

I have discovered a problem of retrieving data from the RuuviTag when the newest firmware version is used. The details are here [1]. Again, I am not exactly sure what causes this problem, but I did not observe it when the older firmware version was used.

[1] - https://f.ruuvi.com/t/getting-data-at-higher-sampling-rate-with-ruuvitag-and-raspberry-pi/1242

ttu, 2018-10-22:
Hi,

Which version were you using? Does log-file have any errors etc. or what is the last info that is in the file before get sensors hangs.

fommike, 2018-10-23:
I used the most recent firmware version of RuuviTag (2.3.0), the most recent ruuvitag-sensor library (0.11.0, python3). I did not get any messages in the log file, the callback result contained a single string, i.e. values from the first and single packet the system captured before hanged.

ttu, 2018-11-21:
@fommike Sorry for a long silence :(

Could you try if you can reproduce the error with the implementation that uses Bleson-library from the brach https://github.com/ttu/ruuvitag-sensor/tree/bleson-ble-communication?

gonecrazy25, 2019-05-14:
I'm having this same issue, when you run find it shows up:

`Start receiving broadcasts (device hci0)
FA:9C:27:9F:90:5E
{'data_format': 4, 'temperature': 21.0, 'humidity': 46.0, 'pressure': 877.0, 'identifier': 'w'}`

but when you run ruuvitag_sensor -g FA:9C:27:9F:90:5E

it just does: `Start receiving broadcasts (device hci0)`
and nothing happens.

I tried with the new bleson branch and it does the same thing with some warnings.

find:
`Finding RuuviTags. Stop with Ctrl+C.
2019-05-14 11:28:22,623   INFO -                 ruuvi.py: 76 -           find_ruuvitags(): Finding RuuviTags. Stop with Ctrl+C.
Start receiving broadcasts (device 0)
2019-05-14 11:28:22,624   INFO -     ble_communication.py:169 -                    start(): Start receiving broadcasts (device 0)
FA:9C:27:9F:90:5E
2019-05-14 11:28:27,447   INFO -                 ruuvi.py: 83 -           find_ruuvitags(): FA:9C:27:9F:90:5E
{'data_format': 4, 'temperature': 21.0, 'humidity': 46.0, 'pressure': 877.0, 'identifier': 'w'}
2019-05-14 11:28:27,447   INFO -                 ruuvi.py: 84 -           find_ruuvitags(): {'data_format': 4, 'temperature': 21.0, 'humidity': 46.0, 'pressure': 877.0, 'identifier': 'w'}
2019-05-14 11:28:30,129 WARNING -       type_converters.py:197 -          from_hcipayload(): TODO: Unhandled GAP type, pos=13 type=0x02 len=3
2019-05-14 11:28:30,129 WARNING -       type_converters.py:198 -          from_hcipayload(): Data: 01 00 01 89 dc f8 b7 98 cd 0e 02 01 06 03 02 af fe 06 09 4e 30 31 36 35 
2019-05-14 11:28:30,129 WARNING -       type_converters.py:199 -          from_hcipayload(): POS : 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 `

and then the get:
`Start receiving broadcasts (device 0)
2019-05-14 11:28:47,032   INFO -     ble_communication.py:169 -                    start(): Start receiving broadcasts (device 0)
2019-05-14 11:28:53,354 WARNING -       type_converters.py:197 -          from_hcipayload(): TODO: Unhandled GAP type, pos=13 type=0x02 len=3
2019-05-14 11:28:53,354 WARNING -       type_converters.py:198 -          from_hcipayload(): Data: 01 00 01 89 dc f8 b7 98 cd 0e 02 01 06 03 02 af fe 06 09 4e 30 31 36 35 
2019-05-14 11:28:53,354 WARNING -       type_converters.py:199 -          from_hcipayload(): POS : 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 `

not sure what to try next.

ttu, 2019-05-14:
Warning in the bleson branch comes from the Bleson package
https://github.com/TheCellule/python-bleson/blob/710a355ae28333d6b760511b711e613e3414efad/bleson/core/hci/type_converters.py#L197

@ojousima any ideas? Is there some data coming from the RuuviTag that is not handled correctly?

ojousima, 2019-05-15:
If I recall correctly, the 2.3.0 sent in connectable advertisement types. This might be the cause of Unhandled GAP type. Please update to 2.4.2 available at lab.ruuvi.com/dfu which is unconnectable and check if the issue remains

ttu, 2019-06-23:
Added a fix for decoder and likely it would fix hanging. On side note it will also blacklist whole sensor as sensor with that firmware sends data that library can't understand. Will make changes if that functionality comes part of released (supported) firmware.

Fixes in commits: https://github.com/ttu/ruuvitag-sensor/commit/352e0d72a54d9dde803b6e58a30c87c54a12ac72 and https://github.com/ttu/ruuvitag-sensor/commit/f6e62125e9021d0977e8f0d6032f4e74a5a9fed8

closed 2019-06-23 by ttu