ttu/ruuvitag-sensor #59: Example tag_test.py is wrong — no output

bevolz, 2018-11-20:
I had to change the tag_test.py script to get any output. Had to change it to:

Import ruuvitag_sensor
From ruuvitag_sensor.log import log
From ruuvitag_sensor.ruuvi import RuuviTagSensor

ruuvitag_sensor.log.enable_console()

RuuviTagSensor.find_ruuvitags()

(The import & from should be lower case).

ttu, 2018-11-21:
Hi @bevolz!

Where is this tag_test.py script?

bevolz, 2018-11-21:
Hi:

It is at https://github.com/ttu/ruuvitag-sensor/blob/master/install_guide_pi.md. Sorry for not including the link.

---
Open nano editor and create test script tag_test.py

$ nano tag_test.py
Add this content. Exit with Ctrl-X and select Y to save

from ruuvitag_sensor.ruuvi import RuuviTagSensor

RuuviTagSensor.find_ruuvitags()
Execute script just created. Expect the same result as from find all tags executed from the command line.

$ python3 tag_test.py
---

  *   Bernie

ttu, 2018-11-22:
Thanks! Didn't remember that there was an example in the install guide :)

Would you like to fix that and do a PR?

bevolz, 2018-11-25:
Hi ... OK, will do.

- Bernie

closed 2018-11-26 by ttu