ttu/ruuvitag-sensor #61: InfluxDB missing fields

lausvi, 2018-12-11:
Hi.

I am using post_to_influxdb.py and it's working otherwise nicely to my needs. However when I do fing_tags I can see the battery values from the RuuviTags, but those are not exported into my InfluxDB. The fields I see there (and can use in my Grafana) are: accelerationX, -Y and -Z, humidity, movementCounter, pressure and temperature.

Using RuuviCollector I can get all those into my InfluxDB, but I'd prefer your Python scripts :)

The tags are brand new, have the default firmware (Weather Station, I believe) and are in the default RAW mode (red led flashing).

ojousima, 2018-12-13:
Thanks for the report, I'll look into this over the weekend

ojousima, 2018-12-20:
Sorry for the delay. 
@lausvi can you check #63 out?

lausvi, 2018-12-20:
No problem!

Just tested it and now it works. Thanks!

PR #63 (2018-12-20): Fix batteryVoltage on Influx example. Use 'ruuvi' as database name
* Fixes #61 
* Fixes battery voltage being placed into measurement counter
* DB name changed into "Ruuvi" on post_to_influxdb_rx.py
* Tested on RPi3 / Python 2
* Verification.py passes
closed 2018-12-20 by lausvi