ttu/ruuvitag-sensor #65: No data getting from sensor

DeDa22, 2019-01-04:
Hi Guys
I'm just finish install ruuvitag-sensor on my raspberry pi following your instruction. 
And finaly when i type ruuvitag -f , this i s the output:

Finding RuuviTags. Stop with Ctrl+C.
Start receiving broadcasts (device hci0)
End Of File (EOF). Exception style platform.
Stop receiving broadcasts

this is my specs in raspberry
Linux raspberrypi 4.14.79
hci0:	Type: Primary  Bus: UART
	BD Address: XX:XX:XX:XX:XX:XX  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:1548 acl:0 sco:0 events:92 errors:0
	TX bytes:2564 acl:0 sco:0 commands:90 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'raspberrypi'
	Class: 0x000000
	Service Classes: Unspecified
	Device Class: Miscellaneous, 
	HCI Version: 4.1 (0x7)  Revision: 0x168
	LMP Version: 4.1 (0x7)  Subversion: 0x2209
	Manufacturer: Broadcom Corporation (15)
Python Version 3.5
pip3 instaled and updated

thank You for answers
with regards
Boris K.

DeDa22, 2019-01-04:
Solved
sudo python3.5 ~/.local/lib/python3.5/site-packages/ruuvitag_sensor -f
sorry for bother you wit such a stupid issue

closed 2019-01-04 by DeDa22

mjazwiecki, 2020-04-25:
i have the same situation - I have followed the instructions in here https://github.com/ttu/ruuvitag-sensor/blob/master/install_guide_pi.md
```
$ ruuvitag_sensor -f
Finding RuuviTags. Stop with Ctrl+C.
Start receiving broadcasts (device hci0)
FYI: Calling a process with sudo!
FYI: Spawning 2 processes with sudo!
End Of File (EOF). Exception style platform.
Stop receiving broadcasts
```

```
$ hciconfig
hci0:   Type: Primary  Bus: UART
        BD Address: XX:XX:XX:XX:XX:XX  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING 
        RX bytes:4592 acl:26 sco:0 events:260 errors:0
        TX bytes:5217 acl:6 sco:0 commands:244 errors:0
```

Linux raspberrypi 4.19.97-v7l+

ttu, 2020-04-25:
Hi @mjazwiecki!

Can you try with ’sudo ruuvitag_sensor -f’?

mjazwiecki, 2020-04-26:
got it working finally - it seems that it should be first installed and then executed by root, which mentioned page did a poor job explaining. thanks!

ttu, 2020-04-26:
Good that you got it working. Would you like to update the guide and make a pull request?

mjazwiecki, 2020-04-26:
sure, done!