ttu/ruuvitag-sensor #67: Decoder index out of range

quulah, 2019-02-12:
An exception jumped out of the decoder:

```
2019-02-11 14:30:41,825  ERROR -               decoder.py:161 -              decode_data(): Value: b'\x03' not valid
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/ruuvitag_sensor/decoder.py", line 148, in decode_data
    acc_x, acc_y, acc_z = self._get_acceleration(byte_data)
  File "/usr/local/lib/python3.6/dist-packages/ruuvitag_sensor/decoder.py", line 130, in _get_acceleration
    acc_x = twos_complement((data[6] << 8) + data[7], 16)
IndexError: index out of range
```

quulah, 2019-02-12:
It may be a bit hard to tell what's wrong based on that. One interesting thing is that my own code's debug says that it last got data at `2019-02-11 08:38:31,679`. I don't know what has been done for 6 hours. :)

I also intermittently get these:
```
2019-02-11 15:14:15,695 WARNING -       type_converters.py:193 -          from_hcipayload(): TODO: Unhandled GAP type, pos=35 type=0x0a len=2
2019-02-11 15:14:15,696 WARNING -       type_converters.py:194 -          from_hcipayload(): Data: 01 00 00 e2 e7 86 a1 41 2c 1c 02 01 19 07 03 be fe 26 fe 03 fe 0d ff 01 09 51 12 d7 13 ae bb 24 8c c2 bf 02 0a f6 
2019-02-11 15:14:15,697 WARNING -       type_converters.py:195 -          from_hcipayload(): POS : 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35 36 37 
2019-02-11 15:14:17,339 WARNING -       type_converters.py:193 -          from_hcipayload(): TODO: Unhandled GAP type, pos=35 type=0x0a len=2
2019-02-11 15:14:17,340 WARNING -       type_converters.py:194 -          from_hcipayload(): Data: 01 00 00 e2 e7 86 a1 41 2c 1c 02 01 19 07 03 be fe 26 fe 03 fe 0d ff 01 09 51 12 d7 13 ae bb 24 8c c2 bf 02 0a f6 
```
I'm not sure if they are related.

ojousima, 2019-02-12:
I've noticed that running 2 BLE programs at the same time can cause corrupted data, for example if I run Noble (NodeJS BLE) and ruuvitag-sensor at the same time there's a load of hard-to-replicate bugs on both programs. Might be related?

quulah, 2019-02-13:
In my case, ruuvitag-sensor is the only BLE application running. However, I'm using the ruuvi_rx reactive wrapper implementation.

From the lack of data, looks like it has died again in the night. I'll take a look at it when I get back home and try to wrap things in a way that it doesn't stop completely or restarts when bad things happen.

ttu, 2019-02-14:
I was checking the code and when there is something wrong with the decoding it should stop handling next datas. 

On error decode return none
https://github.com/ttu/ruuvitag-sensor/blob/master/ruuvitag_sensor/decoder.py#L162

And it is just then ignored
https://github.com/ttu/ruuvitag-sensor/blob/master/ruuvitag_sensor/ruuvi.py#L167

I'll try to update a new version of ruuvitag package this weekend and also merge changes to bleson branch and update it to use the latest version of Bleson and do related code changes.

ttu, 2019-02-15:
Version 0.12.0 released, code merged to bleson branch and bleson updated from 0.0.13 to version 0.0.15

@quulah Have you tried with "normal" Bluez version?

quulah, 2019-02-16:
Yes, I started with that, but had trouble with it since I run this in a Docker container. I think it was related to the usage of `sudo` with the `hcitool` commands. I could've probably put sudo in the container, but I tested the bleson branch and it worked, so I haven't looked back since. :)

ttu, 2019-02-17:
@quulah Thats nice. I really hope that Python will eventually get some working cross platform bluetooth solution (and will be extra happy if it will be Bleson).

Little bit off topic, but could also add Dockerfile to the repository. Did you have some extra settings for the image, or just some basic like this?

```docker
FROM python:3

ADD . /ruuvi
WORKDIR /ruuvi

RUN pip install -e .

CMD [ "python", "ruuvitag_sensor", "-s" ]
```

NOTE: I updated setup.py as process-dependecy-links was removed from pip

quulah, 2019-02-17:
I can try to whip up a Dockerfile in a pull request for this repository. And yeah, basically that's it for the Docker image. I'm putting in my own code too and ruuvitag-sensor is in a requirements.txt that gets read by pip.

I'm using `arm32v7/ubuntu:bionic` as the base, as Ubuntu happened to have a version of Python compiled with `socket.AF_BLUETOOTH` support for bleson.

I took a look at the code, as well as the changes, and looks like the merge didn't really change anything in the decoder.py. And as you said, that exception should have been grabbed.

I'm currently running the newer version, and there hasn't been any problems yet so I'll close this.

closed 2019-02-17 by quulah

ttu, 2019-02-19:
@quulah great if you have time to do the Dockerfile 👍 

I should also gather some feedback how this Bleson version is working and if there is no complains I could release a new version to pypi from this branch. Will open a new issue for that.