ttu/ruuvitag-sensor #68: Reactive exception handling

quulah, 2019-03-15:
This is related to #67. I'm still getting some exceptions out of the library code.

I tried adding a catch-all exceptions block, but that doesn't work as the reactive code seems to be using threading and the exceptions happen in another thread.

quulah, 2019-03-15:
I switched to the non-reactive version: `RuuviTagSensor().get_data_for_sensors()` and now it seems to be leaking memory somewhere. I have yet to find the cause.

ttu, 2019-03-17:
`get_data_for_sensors` calls `RuuviTagSensor._get_ruuvitag_datas` which calls `ble.get_datas`.

`BleCommunicationBleson.get_datas` crates a new _Bleson Observer_ in `BleCommunicationBleson.start` [L175](https://github.com/ttu/ruuvitag-sensor/blob/bleson-ble-communication/ruuvitag_sensor/ble_communication.py#L175)

When _GeneratorExit_ is catched (by default after 5 seconds) `BleCommunicationBleson.stop` is called and _Observer_ is stopped. It is possible that _Observer_ is not cleanup up correctly and is leaking memory.

ttu, 2019-03-17:
__RuuviTagReactive__ executes `RuuviTagReactive._run_get_data_background` in own process. It calls `RuuviTagSensor.get_datas`. Maybe wrapping that to try catch and logging error would catch the error?

ttu, 2019-04-24:
I guess Observer is not cleaned up properly, so it leaks memory.
https://github.com/ttu/ruuvitag-sensor/blob/bleson-ble-communication/ruuvitag_sensor/ble_communication.py#L183

Can @WayneKeenan give some tips how cleanup should be done? Should del be called for the adapter?

ttu, 2019-06-22:
Decoder got some bug fixes and those errors shouldn't appear anymore
https://github.com/ttu/ruuvitag-sensor/commit/352e0d72a54d9dde803b6e58a30c87c54a12ac72 and https://github.com/ttu/ruuvitag-sensor/commit/f6e62125e9021d0977e8f0d6032f4e74a5a9fed8

Bleson memory leak fixed with: https://github.com/ttu/ruuvitag-sensor/commit/3093f467230b70c2790a2049fab8e79acaa94a2e

This should be fixed now. If you find more bugs from bleson-branch or have any comments about functionality, I would appreciate comments to [bleson issue](https://github.com/ttu/ruuvitag-sensor/issues/78). Thanks!

closed 2019-06-22 by ttu