ttu/ruuvitag-sensor #69: Running post_to_influxdb.py for a long period creates hundreds of zombie hcitool processes

zbrozek, 2019-04-24:
It also stops actually capturing and uploading data once this starts happening.

`root     28948  0.0  0.3   7220  3216 pts/1    S+   Apr19   0:00 sudo -n hcitool lescan --duplicates
root     28955  0.1  0.0   2476   632 pts/1    S+   Apr19  11:05 hcitool lescan --duplicates
root     29302  0.0  0.3   7220  3420 pts/1    S+   Apr19   0:00 sudo -n hcitool lescan --duplicates
root     29308  0.1  0.0   2476   624 pts/1    S+   Apr19  11:22 hcitool lescan --duplicates
root     29384  0.0  0.3   7220  3332 pts/1    S+   Apr20   0:00 sudo -n hcitool lescan --duplicates
root     29391  0.1  0.0   2476   648 pts/1    S+   Apr20  10:29 hcitool lescan --duplicates
root     29406  0.0  0.3   7220  3332 pts/1    S+   Apr20   0:00 sudo -n hcitool lescan --duplicates
root     29413  0.1  0.0   2476   616 pts/1    S+   Apr20   8:43 hcitool lescan --duplicates
root     29642  0.0  0.3   7220  3416 pts/1    S+   Apr19   0:00 sudo -n hcitool lescan --duplicates
root     29649  0.1  0.0   2476   624 pts/1    S+   Apr19  11:27 hcitool lescan --duplicates
root     30432  0.0  0.3   7220  3416 pts/1    S+   Apr20   0:00 sudo -n hcitool lescan --duplicates
root     30438  0.1  0.0   2476   668 pts/1    S+   Apr20   9:08 hcitool lescan --duplicates
root     30466  0.0  0.3   7220  3216 pts/1    S+   Apr20   0:00 sudo -n hcitool lescan --duplicates
root     30473  0.1  0.0   2476   656 pts/1    S+   Apr20   8:37 hcitool lescan --duplicates
root     30576  0.0  0.3   7220  3320 pts/1    S+   Apr20   0:00 sudo -n hcitool lescan --duplicates
root     30583  0.1  0.0   2476   624 pts/1    S+   Apr20   9:12 hcitool lescan --duplicates
root     30700  0.0  0.3   7220  3420 pts/1    S+   Apr20   0:00 sudo -n hcitool lescan --duplicates
root     30707  0.1  0.0   2476   636 pts/1    S+   Apr20   9:02 hcitool lescan --duplicates
root     31163  0.0  0.3   7220  3436 pts/1    S+   Apr19   0:00 sudo -n hcitool lescan --duplicates
root     31171  0.1  0.0   2476   656 pts/1    S+   Apr19  11:03 hcitool lescan --duplicates
root     31322  0.0  0.3   7220  3348 pts/1    S+   Apr20   0:00 sudo -n hcitool lescan --duplicates
root     31329  0.1  0.0   2476   616 pts/1    S+   Apr20  10:39 hcitool lescan --duplicates
root     31853  0.0  0.3   7220  3436 pts/1    S+   Apr23   0:00 sudo -n hcitool lescan --duplicates
root     31860  0.0  0.0   2476   624 pts/1    S+   Apr23   0:14 hcitool lescan --duplicates
root     31930  0.0  0.3   7220  3320 pts/1    S+   Apr21   0:00 sudo -n hcitool lescan --duplicates
root     31935  0.1  0.0   2476   620 pts/1    S+   Apr21   8:08 hcitool lescan --duplicates
root     31998  0.0  0.3   7220  3204 pts/1    S+   Apr23   0:00 sudo -n hcitool lescan --duplicates
root     32005  0.1  0.0   2476   668 pts/1    S+   Apr23   2:10 hcitool lescan --duplicates
root     32276  0.0  0.3   7220  3364 pts/1    S+   Apr21   0:00 sudo -n hcitool lescan --duplicates
root     32283  0.1  0.0   2476   620 pts/1    S+   Apr21   6:07 hcitool lescan --duplicates
root     32712  0.0  0.3   7220  3340 pts/1    S+   Apr21   0:00 sudo -n hcitool lescan --duplicates
root     32718  0.1  0.0   2476   660 pts/1    S+   Apr21   8:25 hcitool lescan --duplicates`

ttu, 2019-04-24:
There must be something wrong with killing hcitool process
https://github.com/ttu/ruuvitag-sensor/blob/master/ruuvitag_sensor/ble_communication.py#L90

Any ideas?

zbrozek, 2019-05-04:
Possibly? But how would I find out? It doesn't look like errors are propagated anywhere. Maybe the stored PID is wrong?

ttu, 2019-05-22:
Will make a fix for hcitool process killing

ttu, 2019-05-30:
Fixed hcitool subprocess closing (https://github.com/ttu/ruuvitag-sensor/commit/5deed320ed3661440a04d218e0ccd85b735d14a3).

Also updated the https://github.com/ttu/ruuvitag-sensor/blob/master/examples/post_to_influxdb.py code. For long running processes should use `RuuvitagSensors.get_datas`-method.

closed 2019-05-30 by ttu