ttu/ruuvitag-sensor #87: Server Error when accessing the example HTTP server

richardklingler, 2019-10-28:
When I want to access the example http ruuvi server running a raspberry pi 3b I get a server error 500 as soon I want to access either /data or /data/MAC:

 * Serving Flask app "ruuvi" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
 * Running on http://0.0.0.0:5000/ (Press CTRL+C to quit)
[2019-10-28 12:20:39,879] ERROR in app: Exception on /data [GET]
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/flask/app.py", line 2446, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/.local/lib/python3.7/site-packages/flask/app.py", line 1951, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/.local/lib/python3.7/site-packages/flask/app.py", line 1820, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/.local/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/.local/lib/python3.7/site-packages/flask/app.py", line 1949, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/.local/lib/python3.7/site-packages/flask/app.py", line 1935, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "ruuvi.py", line 55, in get_all_data
    update_data()
  File "ruuvi.py", line 50, in update_data
    allData[key][0] = value
TypeError: tuple indices must be integers or slices, not str
10.0.60.1 - - [28/Oct/2019 12:20:39] "GET /data HTTP/1.1" 500 -

OS: Linux raspberrypi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
Ruuvitag package: ruuvitag_sensor-0.13.0

ttu, 2019-10-29:
Hi. Can you send me your server file (ruuvi.py).

```
File "ruuvi.py", line 50, in update_data
allData[key][0] = value
TypeError: tuple indices must be integers or slices, not str
```

It seems that it uses tuple instead of dictionary for some reason. Couldn't find same line from the default example implementation (https://github.com/ttu/ruuvitag-sensor/blob/master/examples/http_server.py).

richardklingler, 2019-10-29:
Huomenta (o;

Well it is actually this file just with modified MAC addresses:

"""
Simple http server, that returns data in json.
Executes get data for sensors in the background.
Endpoints:
    http://0.0.0.0:5000/data
    http://0.0.0.0:5000/data/<mac>
Requires:
    Flask - pip install flask
"""

from datetime import datetime
import json
from multiprocessing import Manager
from concurrent.futures import ProcessPoolExecutor
from flask import Flask, abort
from ruuvitag_sensor.ruuvi import RuuviTagSensor

app = Flask(__name__)

m = Manager()
q = m.Queue()

allData = {}

tags = {
    'EF:DE:E1:49:3C:9B': 'office',
    'E9:39:41:56:D6:2D': 'livingroom'
}

def run_get_data_background(macs, queue):
    """
    Background process from RuuviTag Sensors
    """
    def callback(data):
        data[1]['time'] = str(datetime.now())
        queue.put(data)

    RuuviTagSensor.get_datas(callback, macs)

def update_data():
    """
    Update data sent by background process to global allData
    """
    global allData
    while not q.empty():
        allData = q.get()
    for key, value in tags.items():
        if key in allData:
            allData[key]['name'] = value

@app.route('/data')
def get_all_data():
    update_data()
    return json.dumps(allData)

@app.route('/data/<mac>')
def get_data(mac):
    update_data()
    if mac not in allData:
        abort(404)
    return json.dumps(allData[mac])

if __name__ == '__main__':
    # Start background process
    executor = ProcessPoolExecutor(1)
    executor.submit(run_get_data_background, list(tags.keys()), q)

    # Strt Flask application
    app.run(host='0.0.0.0', port=5000)

Just redownloaded it and changed the MAC addresses....still the same error:

PR #88 (2019-11-10): Fix http_server example
update_data was overwriting allData with a tuple instead of storing the data into the dict. Just copied the update method from the async http server example.
Fixes #87
closed 2019-11-10 by ttu