ttu/ruuvitag-sensor #89: support RuuvitagSensor first generations ruuvitags? I got an error

Swamplander, 2019-12-09:
Hi,
I have 1 working ruuvitag which are from early kickstarter time (FCC ID 2AKDDRUUVITAG), there is only temperature sensor inside.   

I tried to find ruuvitags with RuuvitagSensor via Python (Windows) and I got next log:

MAC DU:MM:YD:AT:A9:3D
{'data_format': 2, 'temperature': 24.0, 'humidity': 30.0, 'pressure': 995.0, 'identifier': None}
MAC NO:TS:UP:PO:RT:ED
{'data_format': 2, 'temperature': 24.0, 'humidity': 30.0, 'pressure': 995.0, 'identifier': None}
Process finished with exit code 0

Is these some default values from RuuvitagSensor? Should this work with this early generation Ruuvitag as well? However I can't flash Weather station to my ruuvitag:( 

I updated firmware version to 1.2.12 succesfully.

This ruuvitag works with ruuvi Station showing only temperature.

ttu, 2019-12-10:
Hi. Unfortunately this package only supports Linux. Windows and Mac support are only for testing data decoding.

I have been waiting for a working cross platform BLE library for Python, so could change the implementation to use that, but haven't found one yet.

closed 2019-12-10 by ttu

Swamplander, 2019-12-10:
Thanks for information! Kiitos tiedosta, täytynee sitten ottaa DIY projektiksi Raspberry pi, jonka kautta homman pitäisi toimia. Olisi ollut vain itselleni näppärä saada Windowsista toimimaan, kun tässä on asennettuna python IDE.  Harmillista, jos ei ole hyvää BLE kirjastoa tarjolla Windows maailmaan.

ttu, 2019-12-10:
Itse olen käyttänyt samanlaista ympäristöä (Win + Pi). Kannattaa katsoa remote developement ssh:n yli esim.
´https://www.hanselman.com/blog/VisualStudioCodeRemoteDevelopmentOverSSHToARaspberryPiIsButter.aspx

Tai sitten vaan kehittää Windowsilla, kopioida uudet koodit pi:lle ja katsoa mihin kaatuu seuraavaksi :)

Swamplander, 2019-12-11:
Kiitos vinkistä:) Tuohan kuulostaa näppärältä:) Kyllähän minun koodit joka tapauksessa kaatuu, mutta jäisi välivaihe pois, jos koodit voisi siirtää remotella:)