ttu/ruuvitag-sensor #95: Mac based data collection (-g) does not work.

ikke-t, 2020-02-22:
**Describe the bug**

I can list device (-f) and stream the data using (-s). However I don't succeed to get any data out using -g <MAC>. I have tried using both lower case and capital letters in mac, and as well without colons. For the latter one it actually complains about not to be right.

What is required to specify the mac for -g? Do I miss some package from my linux?

It's not like I yet have more than one device for testing, but if I buy more of these, I'd like to select the device I listen to. I didn't modify this software, I just run ruuvitag-sensor with -f / -s / -g options.

**Environment (please complete the following information):**
 - OS: Fedora 29
 - ruuvitag_sensor package version: ruuvitag-sensor-0.13.0
 - RuuviTag firmware version: I don't know where to see it, I updated it couple of weeks ago, so very recent with no modifications by me.
 - Data format 'data_format': 5

ikke-t, 2020-02-22:
If I use the python example, it works however:

```
from ruuvitag_sensor.ruuvitag import RuuviTag

sensor = RuuviTag('C3:47:44:62:7A:5F')

# update state from the device
state = sensor.update()

# get latest state (does not get it from the device)
state = sensor.state

print(state)
```

So likely I'm somehow giving the mac address in some wrong format as this won't work?

```
python3 ~/.local/lib/python2.7/site-packages/ruuvitag_sensor --get C3:47:44:62:7A:5F
```

ttu, 2020-02-25:
Hi! 

Unfortunately can't test this myself in a while (BLE device problems). But we can try to figure this out...

MAC doesn't require any specific format and because rest of the commands and the example you used are working, everything should be fine.

If you remove the `state = sensor.state` from your example will the code still work?
```
from ruuvitag_sensor.ruuvitag import RuuviTag

sensor = RuuviTag('C3:47:44:62:7A:5F')

# update state from the device
state = sensor.update()

print(state)
```

That is how the state is printed when ruuvitag_sensor is executed with mac parameter. (https://github.com/ttu/ruuvitag-sensor/blob/master/ruuvitag_sensor/__main__.py#L39)

So it is possible that `state = sensor.update()` doesn't update the state for some reason?

ikke-t, 2020-02-25:
I might have been unclear in my description. What I meant, the ruuvitag module itself doesn't work if called this way:

```
python3 ~/.local/lib/python2.7/site-packages/ruuvitag_sensor --get C3:47:44:62:7A:5F
```

For some reason it won't find anything if used like that.

ttu, 2020-02-25:
I also might have been pretty unclear with my comments 😄 

When module is started with `--get` argument, it will execute this code

```python
if args.mac_address:
        sensor = RuuviTag(args.mac_address, args.bt_device)
        state = sensor.update()
        log.info(state)
```

Which is pretty similar to the code that you tried and which was working

```py
rom ruuvitag_sensor.ruuvitag import RuuviTag

sensor = RuuviTag('C3:47:44:62:7A:5F')

# update state from the device
state = sensor.update()

# get latest state (does not get it from the device)
state = sensor.state

print(state)
```

It is just missing this line

```py
state = sensor.state
```

So if you can try to execute this code, it is almost the same as the code when using the `--get` argument

```py
rom ruuvitag_sensor.ruuvitag import RuuviTag

sensor = RuuviTag('C3:47:44:62:7A:5F')

# update state from the device
state = sensor.update()

print(state)
```

If that is still working, then you can change the line `sensor = RuuviTag('C3:47:44:62:7A:5F')` to `sensor = RuuviTag('C3:47:44:62:7A:5F', '')`

And if that is still working, then you can try to add hyphens around the MAC. Sholdn't make any difference but you never know 😄 
```
python ~/.local/lib/python2.7/site-packages/ruuvitag_sensor --get 'C3:47:44:62:7A:5F'
```

If that is not working, then we have to wait for the next week, when I should get my BLE dev machine back to working state, so then I can do some testing.

ttu, 2020-03-14:
Ok, there was a clear bug in the code :)

Made fix: https://github.com/ttu/ruuvitag-sensor/pull/104

PR #104 (2020-03-14): Fix RuuviTag-object's get data
Fixes #95
closed 2020-03-14 by ttu