ttu/ruuvitag-sensor #97: could sudo be avoided?

ikke-t, 2020-02-23:
**Describe the bug**
Hi,
I just created a user for running ble to mqtt service. It however turns out this code uses sudo. I'm wondering what would it take to get rid of it? Would it be just about finding the right parameters for udev?

I tried this: https://www.raspberrypi.org/forums/viewtopic.php?t=108581

but then run into the reset can not be performed:
```
[ruuvi@ohuska ruuvitag_sensor]$ hciconfig hci0 reset
Can't down device hci0: Operation not permitted (1)
```

Has anyone figured out how to avoid running things as root?

**Environment (please complete the following information):**
 - OS: Fedora 29
 - ruuvitag_sensor package version: 0.13

JukkaKeisala, 2020-08-26:
I also tried to get it working without `sudo` last year. Spent hours on it. Didn't manage. Perhaps one could add the user to correct group (bluetooth?) by `sudo usermod -aG bluetooth $USER`?

gingerlime, 2020-12-21:
Perhaps only loosely related, but I'm having trouble running inside docker (as part of https://github.com/zewelor/bt-mqtt-gateway) and keep getting `Can't down device hci0: Operation not permitted (1)` even though it seems to run as root (I'm running docker as root, and the user id inside docker is also root, running on raspberry pi 3).

Any tips how to resolve this?

JukkaKeisala, 2020-12-21:
See

https://stackoverflow.com/a/64126744

gingerlime, 2020-12-21:
Thank you @JukkaKeisala. using `--privileged` seems to be the key! 👍

perapp, 2020-12-26:
Also an issue when running in Alpine Linux containers as it does in Home Assistant. Here sudo isn't installed.
Can it be checked if already root and then avoid calling sudo?
That would help getting Ruuvi tags up and running in Home Assistant.

JukkaKeisala, 2020-12-26:
`$EUID` environment variable is 0 for root and non-zero for other users. One could use this to check if root or not.

ikke-t, 2020-12-28:
With containers it quickly gets even more complicated. In properly protected setup there are user namespaces in use, which means that the root in the container is mapped to (random) user in host namespace. And one should not normally veven be root in container, which means the user in container is often also random UID. Then there is perhaps the SELinux still protecting the host.

Just to give heads up for anyone doing this from container, root in container often is not root on host. Hopefully this saves some debug time in such case.

https://opensource.com/article/18/12/podman-and-user-namespaces

samuelphy, 2021-01-12:
> Also an issue when running in Alpine Linux containers as it does in Home Assistant. Here sudo isn't installed.
> Can it be checked if already root and then avoid calling sudo?
> That would help getting Ruuvi tags up and running in Home Assistant.

There exist a custom component for Ruuvi Tags for Home Assistant, uses simple-ruuvitag (based on TTU ruuvitag-sensor) [https://github.com/ruuvi-friends/ruuvi-hass.io](https://github.com/ruuvi-friends/ruuvi-hass.io)

perapp, 2021-01-12:
> There exist a custom component for Ruuvi Tags for Home Assistant, uses simple-ruuvitag (based on TTU ruuvitag-sensor) https://github.com/ruuvi-friends/ruuvi-hass.io

Thanks @samuelphy. Yes, I found that as well. I've moved over to that one now and works fine after a few changes I made in PR https://github.com/ruuvi-friends/ruuvi-hass.io/pull/14

ttu, 2021-01-16:
If sudo needs to be avoided, I would recommend to try optional Bleson communication (https://github.com/ttu/ruuvitag-sensor#bleson).

We could also add environment variable e.g. `RUUVI_BLUEZ_USE_SUDO` to check if _sudo_ needs to be used with BlueZ commands, but I would recommend first to check if using Bleson would be a working solution.

ttu, 2022-06-18:
This was resolved with PR https://github.com/ttu/ruuvitag-sensor/pull/143

If the issue still persists, please reopen the PR.

closed 2022-06-18 by ttu