ttu/ruuvitag-sensor #106: [Discussion] Findings when implementing mac os

sergioisidoro, 2020-04-10:
I have been playing with bleson on mac os, so I though I would add here the [🐉 dragons I discovered 🐉](https://en.wikipedia.org/wiki/Here_be_dragons)

* Bleson version currently published in Pypi is greatly outdated. Published version is 0.0.10, while 0.0.13 is the most recent. Edge version is 0.0.16, but it's not appropriately tested due to CI 

   - https://github.com/TheCellule/python-bleson/issues/56
   - https://github.com/TheCellule/python-bleson/pull/67

* MAC OS does not return MAC addresses (🥁) [See thread here](https://forums.developer.apple.com/thread/8442) - Version 5 data format from ruuvi contains the mac in the payload (I guess because they noticed this while building the iOS app), so we cannot rely on the address provided by bleson

I will add more stuff here as I find it, but though it would be nice to share.
In the meantime I

sergioisidoro, 2020-04-10:
`advertisement.mfg_data` returns a `_NSInlineData` in mac os that needs to be explicitly casted to a byte array or a string.

ttu, 2020-04-11:
Added fix for checking MAC from payload advertised MAC is not available and payload contains MAC. https://github.com/ttu/ruuvitag-sensor/pull/107

`mfg_data` needs still a fix.

sergioisidoro, 2020-04-11:
One more thing I found. Right now Bleson supports continuous scanning in mac os, but in linux it removes duplicates, which basically means we only get one sensor measurement per observer. We need to stop and start the observer if we want to see other measurements. 

That should be fixed here: https://github.com/TheCellule/python-bleson/issues/40 
but it is unreleased. 

Thanks, I'll try to test this out

ttu, 2020-04-19:
First I used the Bleson from pip, then installed it from github with tag, but now instructions have just to install it from the master. Maybe should change it to install from last working commit?

```
$ pip install git+https://github.com/TheCellule/python-bleson.git@[commitId]
```

sergioisidoro, 2020-04-20:
@ttu another caveat. The latest tag of bleson is 0.16, but master `setup.py` still has the version as 0.13 so if you install from master you might not get the master code, if you had a `0.13` installed before. 

I made a PR there, but they say it's not ready for releases because of some CI problem
https://github.com/TheCellule/python-bleson/pull/67

You might need to do a bit of forcing `pip install git+https://github.com/TheCellule/python-bleson.git@[commitId]#egg=bleson==[something larger than 0.13]` by specifying the egg version

ttu, 2023-09-09:
Bleson support will be dropped in favour of Bleak communication.

closed 2023-09-09 by ttu