ttu/ruuvitag-sensor #110: getting data using cron

mjazwiecki, 2020-04-26:
i combined scripts you have in example to make one which aims to be executed every 10 minutes, gets 10 readings from the ruuvitag and then stop.
```
#!/usr/bin/python3.7

from influxdb import InfluxDBClient

from ruuvitag_sensor.ruuvi import RuuviTagSensor, RunFlag

client = InfluxDBClient(host='X.X.X.X', port=8086, database='xxxxx')

# List of macs of sensors which will execute callback function
macs = ['XX:XX:XX:XX:XX:XX']

# make ten measurements at a time
counter = 10
# RunFlag for stopping execution at desired time
run_flag = RunFlag()

def write_to_influxdb(received_data):
    mac = received_data[0]
    payload = received_data[1]

    dataFormat = payload['data_format'] if ('data_format' in payload) else None
    fields = {}
    fields['temperature']               = payload['temperature'] if ('temperature' in payload) else None
    fields['humidity']                  = payload['humidity'] if ('humidity' in payload) else None
    fields['pressure']                  = payload['pressure'] if ('pressure' in payload) else None
    fields['accelerationX']             = payload['acceleration_x'] if ('acceleration_x' in payload) else None
    fields['accelerationY']             = payload['acceleration_y'] if ('acceleration_y' in payload) else None
    fields['accelerationZ']             = payload['acceleration_z'] if ('acceleration_z' in payload) else None
    fields['batteryVoltage']            = payload['battery']/1000.0 if ('battery' in payload) else None
    fields['txPower']                   = payload['tx_power'] if ('tx_power' in payload) else None
    fields['movementCounter']           = payload['movement_counter'] if ('movement_counter' in payload) else None
    fields['measurementSequenceNumber'] = payload['measurement_sequence_number'] if ('measurement_sequence_number' in payload) else None
    fields['tagID']                     = payload['tagID'] if ('tagID' in payload) else None
    fields['rssi']                      = payload['rssi'] if ('rssi' in payload) else None
    json_body = [
        {
            'measurement': 'ruuvi_measurements',
            'tags': {
                'mac': mac,
                'dataFormat': dataFormat
            },
            'fields': fields
        }
    ]
    client.write_points(json_body)
    global counter
    counter = counter - 1
    if counter <= 0:
        run_flag.running = False

RuuviTagSensor.get_datas(write_to_influxdb, macs, run_flag)
```
after 
$ chmod +x ./script.py 
and 
$ sudo crontab -e
```
*/10 * * * * /full/path/to/file/script.py >/dev/null 2>&1
```

it still does not execute, even though from command line it works like a charm and data land into influx.

any ideas to move this forward are appreciated.

ttu, 2020-04-26:
Can you enable logging? Maybe from after that we can see what causes the error.

https://github.com/ttu/ruuvitag-sensor#logging-and-print-to-console

mjazwiecki, 2020-04-26:
Ok, it kinda works now, but strangely enough not every-time as per cron schedule. Once I investigated it further, it seems that `get_datas`  is not always able to get data from ruuvitag. Script is in fact executed but no readings are received. It works in what looks like random intervals, which is super weird.

ttu, 2021-04-29:
I will close this issue now. If the problem appears again, please reopen this issue.

closed 2021-04-29 by ttu