ttu/ruuvitag-sensor #114: How do I update my tag to a different data type?

yudhiesh1997, 2020-10-13:
I am trying to identify the sensors near me but I am getting the following error 

```
DATA TYPE 2 IS OBSOLETE. UPDATE YOUR TAG
```
Even though I am using the latest version of 2.5.9.

precabal, 2020-11-18:
same here. getting the same message after updating to 3.28.13

ttu, 2020-11-22:
Hi!

Check the guide how to change data mode from RuuviTag-site: https://lab.ruuvi.com/dfu/

Should be like this:
`The firmware boots up first time in the mode 1. Button "B" changes the mode to 2. On button presses, the modes cycle 1 -> 2 -> 3 -> 1 -> 2 -> 3 -> 1...`

closed 2020-11-22 by ttu

ttu, 2023-03-18:
Hi @MrGeCode ! Can you enable debug logging (https://github.com/ttu/ruuvitag-sensor#log-all-events-to-log-file) and send me lines before the warning message (`DATA TYPE x IS OBSOLETE. UPDATE YOUR TAG`). There should be something starting with `"Parsing advertisement data:`. Thanks!