ttu/ruuvitag-sensor #117: Data not returned with listed MACs

quulah, 2020-11-29:
**Describe the bug**

Given the following test script data is not returned if `macs` parameter is given:

```
from ruuvitag_sensor.ruuvi import RuuviTagSensor

timeout_in_sec = 4

print('No filter')
datas = RuuviTagSensor.get_data_for_sensors([], timeout_in_sec)
print(datas)

print('Filter with MAC')
datas = RuuviTagSensor.get_data_for_sensors(['CE:99:67:53:BB:CC'], timeout_in_sec)
print(datas)
```

->

```
# python3 test.py
No filter
{'CE:99:67:53:B8:CC': {'data_format': 5, 'humidity': 25.2, 'temperature': 29.51, 'pressure': 1018.79, 'acceleration': 1012.4820986071803, 'acceleration_x': -824, 'acceleration_y': 588, 'acceleration_z': 20, 'tx_power': 4, 'battery': 2431, 'movement_counter': 147, 'measurement_sequence_number': 3221, 'mac': 'ce996753b8cc'}}
Filter with MAC
{}
```

I started using this library though the [bt-mqtt-gateway](https://github.com/zewelor/bt-mqtt-gateway), which seems to call `RuuviTag().update()` directly. It returned no data. After some digging, I found that there's something wrong with how the filtering / whitelisting works.

For example, the `find_ruuvitags()` example from the README works, along with `get_datas()` with an empty `macs` list. But if the MACs are given, then nothing comes out.

I went ahead and added some debug prints to the library, but to my eye the MAC comparison should work. Both MAC strings are the same. I tried different Python versions and library versions for good measure, to no avail.

**Environment (please complete the following information):**
 - OS: Python Docker image with Alpine 3.11 or 3.12
 - ruuvitag_sensor package version: 1.0.1 or 1.1.0
 - RuuviTag firmware version: 2.5.9
 - Data format: 5

JukkaKeisala, 2020-11-29:
Different mac? `B8` vs `BB`

quulah, 2020-11-29:
Well... That was stupid.

Thanks! :)

closed 2020-11-29 by quulah