ttu/ruuvitag-sensor #118: mac address case sensitive?

gingerlime, 2020-12-22:
**Describe the bug**
I'm using ruuvitag-sensor via https://github.com/zewelor/bt-mqtt-gateway and it seems like no data is available if the mac address is in lowercase, but works fine if the mac is all capitalized

**Environment (please complete the following information):**
 - OS: Raspbian
 - ruuvitag_sensor package version: [unsure](https://github.com/zewelor/bt-mqtt-gateway/blob/6d43ddb57db1e53b68c48a88ca56c9fd644a40c7/workers/ruuvitag.py)
 - RuuviTag firmware version: not sure how to see on the iOS app...
 - Data format: 5

ttu, 2021-01-03:
Thanks for the report!

Not familiar with gateway’s implementation, but we could change the MAC comparisons to be case insensitive.

https://github.com/ttu/ruuvitag-sensor/blob/4c582282385f835ef999cbcf9a593aa487a7f6f2/ruuvitag_sensor/ruuvi.py#L160

Should be a pretty easy change, so I will label this as help wanted.

gingerlime, 2021-01-03:
@ttu thank you. I created a small PR, although I'm not sure if it really solves the issue properly... see PR comments.

ttu, 2021-08-01:
After doing some research it seems that MAC-address is not a case sensitive, but appears to be always either upper or lower case.

Because of that I think it is better to leave the solution to be as it is now.

Thanks for all the work you did and although we didn't take it to use, at least we learned something about MAC-addresses 😄

closed 2021-08-01 by ttu