ttu/ruuvitag-sensor #122: Python callbacks doesn't work with firmware 3.x.x?

mgrey, 2021-03-11:
I have two identical setups with Raspberry PI4, one with 2.x.x Ruuvis around, one with only 3.x.x Ruuvis.
The production firmware works fine, I get callbacks for all the broadcasts. For the beta firmware I get only one for each Ruuvi, then nothing more.
So, is the issue firmware? If so, is there a way to use firmware 3.x.x with the ruuvitag_sensor package?

The requirement is the 'Weather Station' firmware, I guess this is now the production firmware?

 - OS: Raspberry PI OS 5.10.17-v7l+(PI4)
 - ruuvitag_sensor package version: 1.1.0
 - RuuviTag firmware version: 2.5.9 and 3.28.x and 3.29.x
 - Data format: 5

ttu, 2021-03-14:
Hi,

Unfortunately package doesn't support yet beta firmware versions, but only production version. Hope we can release the updated version of the package soon.

aleksei-panin, 2021-04-13:
Hello, same problem. 

I updated the firmware of my two ruuvitags after reading this official [page](https://ruuvi.com/software-update/). There nothing is said that 3.x.x is actually a beta version and after upgrading to 3.x.x one cannot anymore downgrade to 2.x.x without a wired connection. This info is available only [here](https://lab.ruuvi.com/dfu/)

Please address this issue as soon as you can and also update documentation on the first link with information above, otherwise it is confusing and may bring other people to the situation where I am at the moment.

ojousima, 2021-04-14:
Is there something we can do on Ruuvi's side to help this issue? 
The data format has not changed, but since RuuviTags have a scan reply and are connectable some header bytes might have changed which might cause this package to miss the transmissions

mgrey, 2021-04-14:
This works for both firmware as a workaround:

from ruuvitag_sensor.adapters.nix_hci import BleCommunicationNix
import ruuvitag_sensor.log
from ruuvitag_sensor.data_formats import DataFormats
from ruuvitag_sensor.decoder import get_decoder
import re

ruuvitag_sensor.log.enable_console()

ble = BleCommunicationNix()

for ble_data in ble.get_datas():
   if re.search("FF9904",ble_data[1]):
      (data_format, encoded) = DataFormats.convert_data(ble_data[1])
      sensor_data = get_decoder(data_format).decode_data(encoded)
      print("%s:%s" % (ble_data[0],sensor_data))

aleksei-panin, 2021-04-14:
My workaround for now is to use [get_data_for_sensors](https://github.com/ttu/ruuvitag-sensor#get-data-for-specified-sensors) method inside of an infinite loop with sufficient timeouts. That does the trick for my POC now. Will follow this issue to see when get_datas method with a callback start working on 3.x.x to switch to it.

Thanks for a quick reply anyways!

ojousima, 2021-04-26:
Following up on this, if the root issue was in the scan response of the RuuviTag it is possible (and probably a good idea) to configure the HCITool to use passive scanning which does not request for scan responses. This saves a little bit of battery on the tags :) 

`hcitool lescan --duplicates --passive` on HCITool.

michaelfuckner, 2021-04-28:
Could you please push changes to pypi since all ruuvitags will be shipped with 3.3.0 from next week onwards

ttu, 2021-04-29:
New version is now released to pypi

closed 2021-04-29 by ttu