ttu/ruuvitag-sensor #123: Examples lack entrypoint protection for multiprocessing

dronir, 2021-03-15:
**Describe the bug**
In most of the Python examples, the call to `RuuviTagSensor.get_datas` fails. The fix is simple and suggested by the error message, since this is a common issue with `multiprocessing`. This applies to Windows as well as macOS.

**Environment (please complete the following information):**
 - OS: macOS Catalina
 - ruuvitag_sensor package version: 0.11
 - RuuviTag firmware version: whatever was prodution in early March 2021
 - Data format: 5
 - Bleson
 - Python: 3.9.2
 
**Reproducing**

Running most of the examples just on the command line, the code crashes with a bunch of exceptions on the `get_datas` call. The relevant part of the exception dump is:

```
RuntimeError: 
        An attempt has been made to start a new process before the
        current process has finished its bootstrapping phase.

        This probably means that you are not using fork to start your
        child processes and you have forgotten to use the proper idiom
        in the main module:

            if __name__ == '__main__':
                freeze_support()
                ...

        The "freeze_support()" line can be omitted if the program
        is not going to be frozen to produce an executable.
```

Simply adding the `if __name__ == '__main__':` around the relevant part fixes this crash entirely. In most of the simple examples it's just the single line with the `get_datas` call.

The underlying reason for this is that on Windows and macOS, `multiprocessing` uses the `spawn` method to start a new process, while on Linux it uses `fork`. The [programming guidelines](https://docs.python.org/3/library/multiprocessing.html#multiprocessing-programming) for the package say (under "Safe importing of main module"):

>Make sure that the main module can be safely imported by a new Python interpreter without causing unintended side effects (such a starting a new process).
>…one should protect the “entry point” of the program by using `if __name__ == '__main__':`

ttu, 2021-03-16:
Thanks for the bug report! 

We should change all examples 👍 I'll add help wanted tag.

dronir, 2021-03-18:
I think I'll try to do some soon, while I'm still working on my own code, at least for the examples I can easily try out.

ttu, 2021-03-21:
@dronir great! All commits are appreciated :)

PR #124 (2021-06-11): Add entry point protection to some examples
Fixes #123 for some of the examples. I could not test every example, but at least (on my Macbook) the ones changed in this commit crash before and work after the change.

To summarize: On Windows and macOS, the `multiprocessing` library generally requires that the main code is protected so it is only run once at the main entry point, because on these platforms the default way of making new threads involves essentially `import`ing the code. This means adding `if __name__ == '__main__'` around the main part. Lack of this protection caused the examples  [... cut]
closed 2021-06-11 by ttu