ttu/ruuvitag-sensor #132: Ruuvitags and InfluxDB 2

Natsku123, 2021-07-26:
Hi,

I was following the examples for InfluxDB and tried to adapt them with InfluxDB 2 (which I prefer over 1.x).

I got it working to a point where it received only 1 transmission per RuuviTag.

I noticed that without setting proper dependency requirements to my project, pip installed old versions of both `ruuvitag-sensor` and `influxdb-client` libraries (since they are the only ones that support same dependencies).

The conflicting dependency is `rx`. `influxdb-client` only supports `rx>=3.0.1` down to the oldest version.

Is there any chance that this library would at some point start to support `rx` version 3?

ttu, 2021-07-26:
Hi!

Yes, here is open PR for that I need to finish https://github.com/ttu/ruuvitag-sensor/pull/130

It would be helpful if you could install the updated version straight from GitHub and try it out?
```sh
$ pip install wheel
$ pip install git+https://github.com/ttu/ruuvitag-sensor.git@migrate-rx-to-v3
```

Natsku123, 2021-07-26:
Hi,

I tried it out and well it worked perfectly (at least the last five minutes I've had it running)!

I'll post an update if something comes up!

ttu, 2022-02-19:
@gehwissenlos Hi and sorry for the delay with this update.

There are few examples that need to be updated, but more importantly I need to release 1.2.1 with Python 2 support and then will merge this and release version 2.0.

I updated the `migrate-rx-to-v3`-brach, so you can use that while waiting for v2.0 release.

```sh
$ pip install wheel
$ pip install git+https://github.com/ttu/ruuvitag-sensor.git@migrate-rx-to-v3
```

gehwissenlos, 2022-02-20:
Hi ttu,

thanks for your help. I installed InfluxDB 2.1, this program and executed this workaround. Afterwards I installed the python client ("pip install influxdb-client"). I don't know what this did to rx because it said that the version is too old and it will be uninstalled, but my script is working:

`from influxdb_client import InfluxDBClient
from influxdb_client.client.write_api import SYNCHRONOUS
from ruuvitag_sensor.ruuvi import RuuviTagSensor

token = "tOkEn=="
org = "org1"
bucket = "bucket1"

client = InfluxDBClient(url="http://raspberrypoe:8086", token=token, org=org)
write_api = client.write_api(write_options=SYNCHRONOUS)

def write_to_influxdb(received_data):
    """
    Convert data into RuuviCollector naming schme and scale
    returns:
        Object to be written to InfluxDB
    """
    mac = received_data[0]
    payload = received_data[1]

    dataFormat = payload['data_format'] if ('data_format' in payload) else None
    fields = {}
    fields['temperature']               = payload['temperature'] if ('temperature' in payload) else None
    fields['humidity']                  = payload['humidity'] if ('humidity' in payload) else None
    fields['pressure']                  = payload['pressure'] if ('pressure' in payload) else None
    fields['accelerationX']             = payload['acceleration_x'] if ('acceleration_x' in payload) else None
    fields['accelerationY']             = payload['acceleration_y'] if ('acceleration_y' in payload) else None
    fields['accelerationZ']             = payload['acceleration_z'] if ('acceleration_z' in payload) else None
    fields['batteryVoltage']            = payload['battery']/1000.0 if ('battery' in payload) else None
    fields['txPower']                   = payload['tx_power'] if ('tx_power' in payload) else None
    fields['movementCounter']           = payload['movement_counter'] if ('movement_counter' in payload) else None
    fields['measurementSequenceNumber'] = payload['measurement_sequence_number'] if ('measurement_sequence_number' in payload) else None
    fields['tagID']                     = payload['tagID'] if ('tagID' in payload) else None
    fields['rssi']                      = payload['rssi'] if ('rssi' in payload) else None
    json_body = [
        {
            'measurement': 'ruuvi_measurements',
            'tags': {
                'mac': mac,
                'dataFormat': dataFormat
            },
            'fields': fields
        }
    ]
    #client.write_points(json_body)
    write_api.write(bucket, org, json_body)

if __name__ == "__main__":
    RuuviTagSensor.get_datas(write_to_influxdb)
`

ttu, 2022-08-05:
Version 2.0 supports `rx` version > 3

https://github.com/ttu/ruuvitag-sensor/blob/master/CHANGELOG.md#200---2022-08-03

closed 2022-08-05 by ttu