ttu/ruuvitag-sensor #144: Stuck after "Invalid advertisement data" error.

eRazeri, 2022-01-22:
**Stuck after "Invalid advertisement data" error**
I'm running the RuuviTagSensor.get_datas function inside a container and very randomly I get the following error message:
```
Invalid advertisement data: 1E11079ECADC240EE5A9E004003E2B020100016AE0A6A534E71F0201061BFF99
Traceback (most recent call last):
  File "/root/.local/lib/python3.6/site-packages/ruuvitag_sensor/data_formats.py", line 62, in convert_data
    cdata, data = _dechunk(data)
  File "/root/.local/lib/python3.6/site-packages/ruuvitag_sensor/data_formats.py", line 21, in _dechunk
    raise ValueError("Cannot read %d bytes, data too short: %s" % (dlen, raw))
ValueError: Cannot read 106 bytes, data too short: 6AE0A6A534E71F0201061BFF
```
I have not been able to catch this (I'm not a python coder) and to restart the container, it is stuck until manually restarting.

How to handle this error so the container would be restarted or the get_datas-function loop would continue?

**Environment:**
 - OS: [Balena Debian Python 3 Run]
 - ruuvitag_sensor package version: [PyPI 1.2.0]
 - RuuviTag firmware version: [3.31.1]
 - Data format: [5]
 - Hardware: Raspberry Pi 3 B

Lalufu, 2022-03-04:
That doesn't look like a ruuvitag advertisement (it doesn't look like a valid BLE advertisement at all unless there's some data getting lost somewhere). From the code, this should just cause the sending MAC to be blacklisted, but the process as a whole should move on.
How many tags do you have? Are you losing data from all of them after this?

eRazeri, 2022-03-04:
Thank you for your reply!
I have only one tag in that location.
This was happening before and after upgrading the firmware of the ruuvitag to the latest.
So if it blacklists the MAC, then I understand that it stops to read data from this tag.
For now I just restart the container manually and all seems to work after it.
Could I somehow set the process to restart, exit the process or to crash the container when this happens?

ttu, 2022-03-20:
Hi @eRazeri!

I think easiest way to make the process to exit when specific MAC is blacklisted is to create a custom log handler. This will listen for all log events and will exit when we receive a log message that matches the condition.

closed 2022-03-20 by ttu

eRazeri, 2022-04-06:
Hi @ttu!

Thank you very much!
Sorry for the long delay in answering, it took me a long time to verify because the error is very random.
With your code as base, I was able to catch the error and restart the container.
I did change your code a little to work for me. I'm checking for the "Invalid advertisement data", because I never encountered this "Blacklisting MAC" message in my enviroment.

Lalufu, 2022-04-06:
I'm still confused as to how we get this data in the first place, as it doesn't seem to conform to the specification for BLE messages. Maybe parsing the output from hcitool goes wrong somewhere?
If you can turn up the debug level in your application to `debug` you should get some useful output. Unfortunately it'll be a lot of output, so I'm not sure whatever system you're using can store this.