ttu/ruuvitag-sensor #199: Data Format Mismatch

eyes-for-computers, 2022-10-08:
When running the "Get sensor data synchronously with callback" sample, I receive a dictionary defined as: 

'XX:XX:XX:XX:XX:XX', {'data_format': 5, 'humidity': 63.87, 'temperature': 20.68, 'pressure': 986.1, 'acceleration': 989.5534346360483, 'acceleration_x': -104, 'acceleration_y': -12, 'acceleration_z': 984, 'tx_power': 4, 'battery': 2249, 'movement_counter': 59, 'measurement_sequence_number': 57716, 'mac': 'xxxxxxxxxxxx'}

The github main  defines v5 as:

'CB:B8:33:4C:88:4F': { 'data_format': 5, 'battery': 2.995, 'pressure': 1000.43, 'mac': 'cbb8334c884f', 'measurement_sequence_number': 2467, 'acceleration_z': 1028, 'acceleration': 1028.0389097694697, 'temperature': 22.14, 'acceleration_y': -8, 'acceleration_x': 4, 'humidity': 53.97, 'tx_power': 4, 'movement_counter': 70, 'rssi': -65 }

The key issue for me is the lack of 'rssi'. Am I missing something?

 - OS: Raspbian Stretch
 - RuuviTag firmware version: 3.31.1
 - Data format: 5

terop, 2022-10-09:
Yes, RSSI reporting is not yet release, please check the [changelog](https://github.com/ttu/ruuvitag-sensor/blob/master/CHANGELOG.md). If you want the RSSI information you must install the development version, please check the instructions at https://github.com/ttu/ruuvitag-sensor#installation.

closed 2022-10-09 by eyes-for-computers

ttu, 2022-10-09:
> The situation with information about unreleased features shown in documentation is sub-optimal but I don't know what would be the best way to avoid it.

There are few common ways to handle documentation. 

* Dedicated documentation page, where it is easy to select version. 
* Version specific documentation is in PyPi/npm/etc.
* GitHub's main-branch is the latest released version and then there would be a separate development-branch.

This repo has version-specific documentation in PyPi and main-branch has the latest development version. Works often well enough, especially if releases are pretty frequent, so then there is less chance for confusion.