ttu/ruuvitag-sensor #200: Get sensor data with callback stopped working after new firmware

jarsiv, 2022-10-09:
Hi,

I'm really confused now. I updated the firmware for one tag on today and now it get sensor data only once and then my code will not collect data from that one sensor anymore. Other sensors with older firmware are looping and collecting data fine.

Tag with newest firmware 3.31.1 is ruuvi6 "FE:3C:29:07:96:F4"
Other tags are with older firmware, I think it's 2.5.9 and these are looping fine.

Data format with new firmware and old seems to be exactly the same. I don't really see what cause the issue with new firmware to the code what I'm using.

Any ideas? I assume that this is something simple with my noobie code that I'm not see the issue :)

```
import socket

import json
import pickle
from ruuvitag_sensor.ruuvi import RuuviTagSensor
from ruuvitag_sensor.decoder import UrlDecoder
import time

from ruuvitag_sensor.ruuvi import RuuviTagSensor, RunFlag

counter = 3
counter2 = 3
counter3 = 3
counter4 = 3
counter5 = 3
counter6 = 3
counter7 = 3
# RunFlag for stopping execution at desired time
run_flag = RunFlag()

def handle_data(found_data):
    #print('MAC ' + found_data[0])
    #print(found_data[1])
    global counter
    global counter2
    global counter3
    global counter4
    global counter5
    global counter6
    global counter7
    if found_data[0] == 'CE:E0:82:DB:40:A1':
     ruuvi1 = (found_data[1])
     dump = json.dumps(ruuvi1)
     data = json.loads(dump)
     #print("Data:" + str(data))
     temp = float(data['temperature'])
     humi = float(data['humidity'])
     pres = float(data['pressure'])
     batt = float(data['battery'])
     move = float(data['movement_counter'])
     counter = counter - 1
     print ("1:", (counter))
    if found_data[0] == 'E1:97:86:E6:1B:CD':
     ruuvi2 = (found_data[1])
     dump = json.dumps(ruuvi2)
     data2 = json.loads(dump)
     #print("Data:" + str(data2))
     temp2 = float(data2['temperature'])
     humi2 = float(data2['humidity'])
     pres2 = float(data2['pressure'])
     batt2 = float(data2['battery'])
     move2 = float(data2['movement_counter'])
     counter2 = counter2 - 1
     print ("2:", (counter2))
    if found_data[0] == 'DC:9C:F4:B6:5B:23':
     ruuvi3 = (found_data[1])
     dump = json.dumps(ruuvi3)
     data3 = json.loads(dump)
     #print("Data:" + str(data3))
     temp3 = float(data3['temperature'])
     humi3 = float(data3['humidity'])
     pres3 = float(data3['pressure'])
     batt3 = float(data3['battery'])
     move3 = float(data3['movement_counter'])
     counter3 = counter3 - 1
     print ("3:", (counter3))
    if found_data[0] == 'C9:FB:39:F1:A6:65':
     ruuvi4 = (found_data[1])
     dump = json.dumps(ruuvi4)
     data4 = json.loads(dump)
     #print("Data:" + str(data4))
     temp4 = float(data4['temperature'])
     humi4 = float(data4['humidity'])
     pres4 = float(data4['pressure'])
     batt4 = float(data4['battery'])
     move4 = float(data4['movement_counter'])
     counter4 = counter4 - 1
     print ("4:", (counter4))
    if found_data[0] == 'D9:22:97:F1:84:E0':
     ruuvi5 = (found_data[1])
     dump = json.dumps(ruuvi5)
     data5 = json.loads(dump)
     #print("Data:" + str(data5))
     temp5 = float(data5['temperature'])
     humi5 = float(data5['humidity'])
     pres5 = float(data5['pressure'])
     batt5 = float(data5['battery'])
     move5 = float(data5['movement_counter'])
     counter5 = counter5 - 1
     print ("5:", (counter5))
    if found_data[0] == 'FE:3C:29:07:96:F4':
     ruuvi6 = (found_data[1])
     dump = json.dumps(ruuvi6)
     data6 = json.loads(dump)
     print("Data:" + str(data6))
     temp6 = float(data6['temperature'])
     humi6 = float(data6['humidity'])
     pres6 = float(data6['pressure'])
     batt6 = float(data6['battery'])
     move6 = float(data6['movement_counter'])
     counter6 = counter6 - 1
     print ("6:", (counter6))
    if found_data[0] == 'F9:C4:CB:6D:7E:48':
     ruuvi7 = (found_data[1])
     dump = json.dumps(ruuvi7)
     data7 = json.loads(dump)
     #print("Data:" + str(data7))
     temp7 = float(data7['temperature'])
     humi7 = float(data7['humidity'])
     pres7 = float(data7['pressure'])
     batt7 = float(data7['battery'])
     move7 = float(data7['movement_counter'])
     counter7 = counter7 - 1
     print ("7:", (counter7))
    if counter < 0:
     Mes = str(temp),str(humi),str(pres),str(batt),str(move)
     Mesjoin = (", ".join(Mes))
     print ("message:", Mesjoin)
     #with open('/home/pi/ruuvi/tiedot.txt', 'w') as file_to_write:
      #file_to_write.write(str(Mesjoin))
     counter = 3
    if counter2 < 0:
     Mes2 = str(temp2),str(humi2),str(pres2),str(batt2),str(move2)
     Mesjoin2 = (", ".join(Mes2))
     print ("message:", Mesjoin2)
     with open('/home/pi/ruuvi/tiedot2.txt', 'w') as file_to_write:
      file_to_write.write(str(Mesjoin2))
     counter2 = 3
    if counter3 < 0:
     Mes3 = str(temp3),str(humi3),str(pres3),str(batt3),str(move3)
     Mesjoin3 = (", ".join(Mes3))
     print ("message:", Mesjoin3)
     with open('/home/pi/ruuvi/tiedot3.txt', 'w') as file_to_write:
      file_to_write.write(str(Mesjoin3))
     counter3 = 3
    if counter4 < 0:
     Mes4 = str(temp4),str(humi4),str(pres4),str(batt4),str(move4)
     Mesjoin4 = (", ".join(Mes4))
     print ("message:", Mesjoin4)
     #with open('/home/pi/ruuvi/tiedot4.txt', 'w') as file_to_write:
      #file_to_write.write(str(Mesjoin4))
     counter4 = 3
    if counter5 < 0:
     Mes5 = str(temp5),str(humi5),str(pres5),str(batt5),str(move5)
     Mesjoin5 = (", ".join(Mes5))
     print ("message:", Mesjoin5)
     with open('/home/pi/ruuvi/tiedot5.txt', 'w') as file_to_write:
      file_to_write.write(str(Mesjoin5))
     counter5 = 3
    if counter6 < 0:
     Mes6 = str(temp6),str(humi6),str(pres6),str(batt6),str(move6)
     Mesjoin6 = (", ".join(Mes6))
     print ("message:", Mesjoin6)
     with open('/home/pi/ruuvi/tiedot6.txt', 'w') as file_to_write:
      file_to_write.write(str(Mesjoin6))
     counter6 = 3
    if counter7 < 0:
     Mes7 = str(temp7),str(humi7),str(pres7),str(batt7),str(move7)
     Mesjoin7 = (", ".join(Mes7))
     print ("message:", Mesjoin7)
     #with open('/home/pi/ruuvi/tiedot7.txt', 'w') as file_to_write:
      #file_to_write.write(str(Mesjoin7))
     counter7 = 3

# List of macs of sensors which will execute callback function
macs = ['CE:E0:82:DB:40:A1', 'E1:97:86:E6:1B:CD', 'DC:9C:F4:B6:5B:23', 'C9:FB:39:F1:A6:65', 'D9:22:97:F1:84:E0', 'FE:3C:29:07:96:F4', 'F9:C4:CB:6D:7E:48']

RuuviTagSensor.get_datas(handle_data, macs, run_flag)

```

jarsiv, 2022-10-09:
Result with /ruuvitag_sensor -f

```
FE:3C:29:07:96:F4
{'data_format': 5, 'humidity': 41.08, 'temperature': 21.14, 'pressure': 1002.28, 'acceleration': 1000.1599872020475, 'acceleration_x': 8, 'acceleration_y': -16, 'acceleration_z': 1000, 'tx_power': 4, 'battery': 2844, 'movement_counter': 13, 'measurement_sequence_number': 616, 'mac': 'fe3c290796f4'}
E1:97:86:E6:1B:CD
{'data_format': 5, 'humidity': 38.35, 'temperature': 22.18, 'pressure': 1001.54, 'acceleration': 1046.2428016478775, 'acceleration_x': 1032, 'acceleration_y': -172, 'acceleration_z': -4, 'tx_power': 4, 'battery': 2683, 'movement_counter': 233, 'measurement_sequence_number': 54144, 'mac': 'e19786e61bcd'}
DC:9C:F4:B6:5B:23
{'data_format': 5, 'humidity': 87.18, 'temperature': 7.02, 'pressure': 1002.5, 'acceleration': 1020.6272581114027, 'acceleration_x': 1012, 'acceleration_y': -120, 'acceleration_z': 56, 'tx_power': 4, 'battery': 2767, 'movement_counter': 45, 'measurement_sequence_number': 8986, 'mac': 'dc9cf4b65b23'}
D9:22:97:F1:84:E0
{'data_format': 5, 'humidity': 70.35, 'temperature': 6.69, 'pressure': 1002.29, 'acceleration': 1059.34319273784, 'acceleration_x': 1056, 'acceleration_y': -76, 'acceleration_z': 36, 'tx_power': 4, 'battery': 2707, 'movement_counter': 170, 'measurement_sequence_number': 2671, 'mac': 'd92297f184e0'}
```

jarsiv, 2022-10-09:
Ha :D

It was simple solution. ruuvitag-sensor was out of date :)

This can be closed

ttu, 2022-10-09:
Good that you found the solution  :)

closed 2022-10-09 by ttu