ttu/ruuvitag-sensor #211: Development install fails

SailingGreg, 2023-02-22:
**Describe the bug**
A clear and concise description of what the bug is.

The development install fails with an error and I suspect this is because it is not a package as there is no setup.py in the repo

$ python -m venv .venv
$ source .venv/bin/activate
$ python -m pip install git+https://github.com/ttu/ruuvitag-sensor

pi@piauto:~ $ python3 -m venv .venv
pi@piauto:~ $ source .venv/bin/activate
(.venv) pi@piauto:~ $ python3 -m pip install git+https://github.com/ttu/ruuvitag-sensor

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting git+https://github.com/ttu/ruuvitag-sensor
  Cloning https://github.com/ttu/ruuvitag-sensor to /tmp/pip-req-build-fvxu5lo7
  Installing build dependencies ... done
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/usr/lib/python3.7/tokenize.py", line 447, in open
        buffer = _builtin_open(filename, 'rb')
    FileNotFoundError: [Errno 2] No such file or directory: '/tmp/pip-req-build-fvxu5lo7/setup.py'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-req-build-fvxu5lo7/

**Environment (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04, Raspbian Stretch]
 - ruuvitag_sensor package version: [e.g. 0.11, 0.12, master, bleson-branch]
 - RuuviTag firmware version: [e.g. KS, 1.0.1, 1.2.12, 2.4.2]
 - Data format: [e.g. 2, 3, 4, 5]

The environment is Pi Buster (version 10) and I've been running ruuvitag_sensor 2.0 but wanted to explore Bleak support
Python version is 3.7.3

pi@devhost:~ $ python3 -V
Python 3.7.3

ttu, 2023-02-23:
Hi! Can you try to upgrade pip and see if install works after that? It is possible the pip version doesn't support `pyproject.toml`

```sh
$ python -m venv .venv
$ source .venv/bin/activate
$ python -m pip install --upgrade pip
$ python -m pip install git+https://github.com/ttu/ruuvitag-sensor
```

SailingGreg, 2023-03-01:
Hi, checked the module version and then did an upgrade in the virtual environment which upgraded it from 18.1 to 23.0.1. This has worked and `pyproject.toml` is now recognised.

Thanks for the help

```
pi@piauto:~ $ pip3 -V
pip 18.1 from /usr/lib/python3/dist-packages/pip (python 3.7)
```

```
pi@piauto:~ $ source .venv/bin/activate
(.venv) pi@piauto:~ $ python3 -m pip install --upgrade pip
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pip
  Downloading https://files.pythonhosted.org/packages/07/51/2c0959c5adf988c44d9e1e0d940f5b074516ecc87e96b1af25f59de9ba38/pip-23.0.1-py3-none-any.whl (2.1MB)
    100% |████████████████████████████████| 2.1MB 20kB/s 
Installing collected packages: pip
  Found existing installation: pip 18.1
    Uninstalling pip-18.1:
      Successfully uninstalled pip-18.1
Successfully installed pip-23.0.1
```

SailingGreg, 2023-03-01:
Closing issue

closed 2023-03-01 by SailingGreg