ttu/ruuvitag-sensor #216: streaming  from "get_async_bleak.py"

epifanio, 2023-07-07:
**Describe the bug**
I wanted to stream the output of my ruuvi tasg in a python application- it succesfully  find and connect to the device using the "bleak" example, but it stop printing after a few (6) iterations.

**Environment (please complete the following information):**
 - OS: [Ubuntu 22.04]
 - ruuvitag_sensor package version: [master]
 - RuuviTag firmware version: [3.31.1+default]
 - Data format: [5]

when running the following example:

```python
import asyncio
import os

os.environ["RUUVI_BLE_ADAPTER"] = "bleak"

import ruuvitag_sensor.log
from ruuvitag_sensor.ruuvi import RuuviTagSensor

ruuvitag_sensor.log.enable_console()

async def main():
    async for data in RuuviTagSensor.get_data_async():
        print(data)

if __name__ == "__main__":
    for i in range(5): 
        asyncio.get_event_loop().run_until_complete(main())
```

It succesfully print the following log:

```python
./bleak_ble.py:71: FutureWarning: BLEDevice.rssi is deprecated and will be removed in a future version of Bleak, use AdvertisementData.rssi instead
  data += hex((device.rssi + (1 << 8)) % (1 << 8)).replace("0x", "")
Parsing advertisement data: 1c1bFF990405133348bccaa8ffbcff7c04049e360e00aed67bf8b0cc67c9
Found chunk of type FF: FF990405133348bccaa8ffbcff7c04049e360e00aed67bf8b0cc67
Found candidate FF990405133348bccaa8ffbcff7c04049e360e00aed67bf8b0cc67
('D6:7B:F8:B0:CC:67', {'data_format': 5, 'humidity': 46.55, 'temperature': 24.57, 'pressure': 1018.8, 'acceleration': 1038.6683782613197, 'acceleration_x': -68, 'acceleration_y': -132, 'acceleration_z': 1028, 'tx_power': 4, 'battery': 2865, 'movement_counter': 14, 'measurement_sequence_number': 174, 'mac': 'd67bf8b0cc67', 'rssi': -55})
Parsing advertisement data: 1c1bFF990405132d48b1caa7ffb8ff8003fc9e360f00afd67bf8b0cc67ca
Found chunk of type FF: FF990405132d48b1caa7ffb8ff8003fc9e360f00afd67bf8b0cc67
Found candidate FF990405132d48b1caa7ffb8ff8003fc9e360f00afd67bf8b0cc67
('D6:7B:F8:B0:CC:67', {'data_format': 5, 'humidity': 46.52, 'temperature': 24.55, 'pressure': 1018.79, 'acceleration': 1030.518316188509, 'acceleration_x': -72, 'acceleration_y': -128, 'acceleration_z': 1020, 'tx_power': 4, 'battery': 2865, 'movement_counter': 15, 'measurement_sequence_number': 175, 'mac': 'd67bf8b0cc67', 'rssi': -54})
Parsing advertisement data: 1c1bFF990405132648d0caa7ffb4ff7c03f49e360f00b0d67bf8b0cc67ca
Found chunk of type FF: FF990405132648d0caa7ffb4ff7c03f49e360f00b0d67bf8b0cc67
Found candidate FF990405132648d0caa7ffb4ff7c03f49e360f00b0d67bf8b0cc67
('D6:7B:F8:B0:CC:67', {'data_format': 5, 'humidity': 46.6, 'temperature': 24.51, 'pressure': 1018.79, 'acceleration': 1023.3982606981507, 'acceleration_x': -76, 'acceleration_y': -132, 'acceleration_z': 1012, 'tx_power': 4, 'battery': 2865, 'movement_counter': 15, 'measurement_sequence_number': 176, 'mac': 'd67bf8b0cc67', 'rssi': -54})
Parsing advertisement data: 1c1bFF990405132648d0caa7ffb4ff7c03f49e360f00b0d67bf8b0cc67ca
Found chunk of type FF: FF990405132648d0caa7ffb4ff7c03f49e360f00b0d67bf8b0cc67
Found candidate FF990405132648d0caa7ffb4ff7c03f49e360f00b0d67bf8b0cc67
('D6:7B:F8:B0:CC:67', {'data_format': 5, 'humidity': 46.6, 'temperature': 24.51, 'pressure': 1018.79, 'acceleration': 1023.3982606981507, 'acceleration_x': -76, 'acceleration_y': -132, 'acceleration_z': 1012, 'tx_power': 4, 'battery': 2865, 'movement_counter': 15, 'measurement_sequence_number': 176, 'mac': 'd67bf8b0cc67', 'rssi': -54})
Parsing advertisement data: 1c1bFF990405132348e5caa8ffb8ff8003fc9e360f00b1d67bf8b0cc67ca
Found chunk of type FF: FF990405132348e5caa8ffb8ff8003fc9e360f00b1d67bf8b0cc67
Found candidate FF990405132348e5caa8ffb8ff8003fc9e360f00b1d67bf8b0cc67
('D6:7B:F8:B0:CC:67', {'data_format': 5, 'humidity': 46.65, 'temperature': 24.5, 'pressure': 1018.8, 'acceleration': 1030.518316188509, 'acceleration_x': -72, 'acceleration_y': -128, 'acceleration_z': 1020, 'tx_power': 4, 'battery': 2865, 'movement_counter': 15, 'measurement_sequence_number': 177, 'mac': 'd67bf8b0cc67', 'rssi': -54})
Parsing advertisement data: 1c1bFF990405131d48f4caa8ffb8ff7c03fc9e360f00b2d67bf8b0cc67cc
Found chunk of type FF: FF990405131d48f4caa8ffb8ff7c03fc9e360f00b2d67bf8b0cc67
Found candidate FF990405131d48f4caa8ffb8ff7c03fc9e360f00b2d67bf8b0cc67
('D6:7B:F8:B0:CC:67', {'data_format': 5, 'humidity': 46.69, 'temperature': 24.46, 'pressure': 1018.8, 'acceleration': 1031.0227931525083, 'acceleration_x': -72, 'acceleration_y': -132, 'acceleration_z': 1020, 'tx_power': 4, 'battery': 2865, 'movement_counter': 15, 'measurement_sequence_number': 178, 'mac': 'd67bf8b0cc67', 'rssi': -52})

```

but after then it stopsd printing .. like is hanging/waiting .. is that expected? should I modify the example to get  the data streamed without stop?

Thanks!

ttu, 2023-07-08:
Hi! I think for loop is unnecessary. `run_until_complete` will not exit before `main`-function is completed and `get_data_async` will return data forever.

You can try something like this if you want to exit after you have received 10 broadcasts from sensors.
```py
async def main():
    datas = []
    async for found_data in RuuviTagSensor.get_data_async():
        print(f"MAC: {found_data[0]}")
        print(f"Data: {found_data[1]}")
        datas.push(found_data)
        if len(datas) > 10:
            break

if __name__ == "__main__":
    asyncio.get_event_loop().run_until_complete(main())
```

epifanio, 2023-07-08:
Hi, thanks for your reply - yes the `for loop` was there as a leftover of some debugging.

The problem  I am having is that after 6 reads from the sensor, the data are no longer printed - like the program is waiting but nothings comes in. 

if I stop and restart the script, i get 6 more prints and then it hangs again.

ttu, 2023-07-08:
Can you enable debug logging, maybe it will reveal some helpful info

```py
import logging
import ruuvitag_sensor.log
from ruuvitag_sensor.log import log
from ruuvitag_sensor.ruuvi import RuuviTagSensor

ruuvitag_sensor.log.enable_console()

log.setLevel(logging.DEBUG)

for handler in ruuvitag_sensor.log.log.handlers:
    handler.setLevel(logging.DEBUG)
```

epifanio, 2023-07-09:
Hi, @ttu  - thanks a lot for you feedback - I think the issue is in my Ubuntu installation on the desktop machine - the Ruuvi tag sometimes shows listed and sometime doesn't  .. I tried reinstalling some libraries, but now it doesn't show anymore.
I tried the ruuvitag=sensor code on my laptop - whee the ruuvitag shows listed in the BT settings .. and the get_data asysnc works fine - quite slow refresh (intermittent it can take 2 sec or 10 sec .. between two data records) but definitely work. 
Thanks for your help! I will try to fix my BT on the desktop machine

closed 2023-09-09 by ttu