ttu/ruuvitag-sensor #223: hcitool not activating le scan

teeks99, 2023-09-08:
**Describe the bug**
When running the cli client in find mode, `python3 -m ruuvitag_sensor -f --debug`, nothing is detected. 

Something isn't working correctly to start the hcitool command, as I can run `sudo btmon` separately and not see any input. However when I run the command `sudo hcitool -i hci0 lescan2 --duplicates --passive` manually in another terminal, the same btmon session shows all the traffic I'm expecting to see, including the messages with the Ruuvi tags. Similar for `sudo hcidump -i hci0 --raw`

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - ruuvitag_sensor package version: master branch - f8458d10f37c080c335366f2b893accaf8a2221a
 - RuuviTag firmware version: ?
 - Data format: 5

## Running with ruuvitag_sensor

Setup:
```
tomkent@lisa:/mnt/fs4/src/ruuvi_scan$ python3 -m venv env                                                               
tomkent@lisa:/mnt/fs4/src/ruuvi_scan$ . env/bin/activate                                                                
(env) tomkent@lisa:/mnt/fs4/src/ruuvi_scan$                                                                             
(env) tomkent@lisa:/mnt/fs4/src/ruuvi_scan$                                                                             
(env) tomkent@lisa:/mnt/fs4/src/ruuvi_scan$                                                                             
(env) tomkent@lisa:/mnt/fs4/src/ruuvi_scan$ python -m pip install git+https://github.com/ttu/ruuvitag-sensor
Collecting git+https://github.com/ttu/ruuvitag-sensor                                                                   
  Cloning https://github.com/ttu/ruuvitag-sensor to /tmp/pip-req-build-44ui9q0s
  Running command git clone --filter=blob:none --quiet https://github.com/ttu/ruuvitag-sensor /tmp/pip-req-build-44ui9q0
s                                                                                                                       
  Resolved https://github.com/ttu/ruuvitag-sensor to commit f8458d10f37c080c335366f2b893accaf8a2221a                    
  Installing build dependencies ... done                                                                                
  Getting requirements to build wheel ... done                                                                          
  Preparing metadata (pyproject.toml) ... done                                                                          
Collecting ptyprocess                                                                                                   
  Using cached ptyprocess-0.7.0-py2.py3-none-any.whl (13 kB)                                                            
Collecting reactivex                                                                                                    
  Using cached reactivex-4.0.4-py3-none-any.whl (217 kB)                                                                
Collecting typing-extensions<5.0.0,>=4.1.1                                                                              
  Using cached typing_extensions-4.7.1-py3-none-any.whl (33 kB)                                                         
Building wheels for collected packages: ruuvitag-sensor
  Building wheel for ruuvitag-sensor (pyproject.toml) ... done                                                          
  Created wheel for ruuvitag-sensor: filename=ruuvitag_sensor-2.1.0-py3-none-any.whl size=30269 sha256=bed4cd7e0d77751be
d417be63ba78276b2e9c230e52b0572f8969aee3a1f4cec             
  Stored in directory: /tmp/pip-ephem-wheel-cache-zh15guyr/wheels/bf/69/aa/b8101e234baf7de1c300d714366f6f1e3b054e76ba413
a97fd                                                                                                                   
Successfully built ruuvitag-sensor                                                                                      
Installing collected packages: ptyprocess, typing-extensions, reactivex, ruuvitag-sensor                                
Successfully installed ptyprocess-0.7.0 reactivex-4.0.4 ruuvitag-sensor-2.1.0 typing-extensions-4.7.1                   
```

Run
```
(env) tomkent@lisa:/mnt/fs4/src/ruuvi_scan$ python3 -m ruuvitag_sensor -f --debug
Finding RuuviTags. Stop with Ctrl+C.
Start receiving broadcasts (device hci0)
FYI: Calling a process with sudo: hciconfig hci0 reset
FYI: Spawning process with sudo: hcitool -i hci0 lescan2 --duplicates --passive
FYI: Spawning process with sudo: hcidump -i hci0 --raw
^CStop receiving broadcasts
```

I *do* see these two subprocesses running while this command is executing:
```
4 S root      756518  756497  0  80   0 -  3490 -      09:21 pts/37   00:00:00 /usr/bin/sudo hcitool -i hci0 lescan2 --duplicates --passive
4 S root      756519  756497  0  80   0 -  3490 -      09:21 pts/38   00:00:00 /usr/bin/sudo hcidump -i hci0 --raw
```

I left it for about 30 seconds before stopping it. During this time, I had `sudo btmon` and `sudo hcidump -i hci0 --raw > hcidump_raw.log` running in other terminals. The output is attached. Both have a whole bunch of stuff happen when the program starts up, but then are silent for the majority of the 30s. 

*  [btmon-python.log](https://github.com/ttu/ruuvitag-sensor/files/12564152/btmon-python.log)
*  [hcidump_raw-python.log](https://github.com/ttu/ruuvitag-sensor/files/12564154/hcidump_raw-python.log)

## Running hcitool by hand

[... 39 lines cut ...]
        Event type: Non connectable undirected - ADV_NONCONN_IND (0x03)                                                 
        Address type: Random (0x01)                                                                                     
        Address: E0:30:B4:06:CE:72 (Static)                                                                             
        Data length: 31                                                                                                 
        Flags: 0x06                                                                                                     
          LE General Discoverable Mode                                                                                  
          BR/EDR Not Supported                                                                                          
        Company: Ruuvi Innovations Ltd. (1177)                                                                          
          Data: 05114a44a1c271fffcffc403fcb6b6789a3be030b406ce72                                                        
        RSSI: -82 dBm (0xae)                                                                                            
> HCI Event: LE Meta Event (0x3e) plen 43                                                        #138 [hci0] 444.819723 
      LE Advertising Report (0x02)                                                                                      
        Num reports: 1                                                                                                  
        Event type: Non connectable undirected - ADV_NONCONN_IND (0x03)                                                 
        Address type: Random (0x01)                                                                                     
        Address: EF:71:22:67:33:06 (Static)                                                                             
        Data length: 31                                                                                                 
        Flags: 0x06                                                                                                     
          LE General Discoverable Mode                                                                                  
          BR/EDR Not Supported                                                                                          
        Company: Ruuvi Innovations Ltd. (1177)                                                                          
          Data: 0510d846ebc2b3fc4cffe4fe907e7640c7b7ef7122673306                                                        
        RSSI: -66 dBm (0xbe)                                                      
<snip>
```

ttu, 2023-09-09:
Thanks for the issue! Unfortunately I do not have any quick solution in mind for this.

You could try beta-version with bleak communication, but for that you have to install development version from github
https://github.com/ttu/ruuvitag-sensor#installation
```sh
python -m pip install git+https://github.com/ttu/ruuvitag-sensor
```
You also need to install bleak separately
https://github.com/ttu/ruuvitag-sensor#Bleak
```sh
$ python -m pip install bleak
```
And enable it with env variable
```sh
$ export RUUVI_BLE_ADAPTER="bleak"
```

CLI works with same commands, but in Python-code you have to use async-methods https://github.com/ttu/ruuvitag-sensor#2-get-sensor-data-asynchronously.

I will inspect your logs and commands more closely and hopefully I can come up with some solution.

teeks99, 2023-09-10:
I played around a bit trying different things to get python to call the hcitool command. It seems like sudo isn't working correctly when done through either of these:

```
import subprocess
import signal
import os
import ptyprocess
import shlex
import time

timeout_sec = 15

command = "sudo hcitool -i hci0 lescan2 --duplicates --passive"

def use_subprocess(command):
    process = subprocess.Popen(command, stdout=subprocess.PIPE, 
                               stderr=subprocess.STDOUT, shell=True,
                               preexec_fn=os.setsid)
    time.sleep(timeout_sec)
    os.killpg(os.getpgid(process.pid), signal.SIGTERM)  # Send the signal to all the process groups
    
    print(process.stdout.read())

def use_pty(command):
    command_args = shlex.split(command)
    p = ptyprocess.PtyProcess.spawn(command_args)

    time.sleep(timeout_sec)

    p.kill(signal.SIGTERM)
    print(p.read(2048))

#use_subprocess(command)
use_pty(command)
```

I had a bit better luck without the hack to Popen with the setsid stuff, but then I couldn't kill the program when I was done.

Either way, it worked smoothy with bleak, that has the huge benefit of not needing any sudo stuff. I guess I'll roll with that!

ttu, 2023-09-11:
Good to hear that bleak is working fine :+1: 

I guess how sudo works with `subprocess/ptyprocess`  is os/environment specific as you are not the first one to have issues, but for most it works without issues.

I released a new version (2.2.0) with async/bleak-support, but bleak still needs to be installed separately in that version. In next release bleak-adapter will be a default with Win & macOS and eventually bleak will replace BlueZ-adapter on Linux as default adapter.
https://github.com/ttu/ruuvitag-sensor/pull/220

closed 2023-09-11 by ttu