ttu/ruuvitag-sensor #233: RSSI data missing with the Bleson adapter

sevesalm, 2023-11-25:
**Describe the bug**

When using the default BlueZ adapter, the RSSI level is included in the data. When using Bleson, it has the value `None`.

Simple sample code shows the issue. 

````python
import sys
import os

os.environ["RUUVI_BLE_ADAPTER"] = "bleson" # Remove this line to make RSSI appear in the data

from ruuvitag_sensor.ruuvi_rx import RuuviTagReactive
from reactivex import operators as ops

ruuvis = RuuviTagReactive(['TODO_MAC']) # Add your MAC here
ruuvi_emissions = ruuvis.get_subject()
sensor_data = ruuvi_emissions.pipe(
        ops.scan(lambda acc, x: acc | {x[0]: x[1]}, {}),
        ops.first()
        ).run()
ruuvis.stop()
print(sensor_data)
sys.exit()
````

**Environment (please complete the following information):**
 - OS: Debian GNU/Linux 11 (bullseye) on Raspberry Pi 3B+
 - ruuvitag_sensor package version: 2.3.0
 - RuuviTag firmware version: 3.31.1+default
 - Data format: 5

ttu, 2023-11-27:
Hi! I don't have a working Linux setup available at the moment and it seems that Bleson doesn't work with my macOS.

Could you enable debug logging and send me what data Bleson has in [Advertisement](https://bleson.readthedocs.io/en/latest/api_datatypes.html#advertisement) which is logged [here](https://github.com/ttu/ruuvitag-sensor/blob/0015efcf43b2879eecd7dc0eda9e993962508bb2/ruuvitag_sensor/adapters/bleson.py#L26).

It is possible that RSSI should be added to our own payload same way as in [Bleak-adapter](https://github.com/ttu/ruuvitag-sensor/blob/0015efcf43b2879eecd7dc0eda9e993962508bb2/ruuvitag_sensor/adapters/bleak_ble.py#L77).

sevesalm, 2023-11-30:
I think this is the correct data:

````
2023-11-30 20:38:11,756 DEBUG -                bleson.py: 26 - _run_get_data_background(): Data: Advertisement(flags=0x06, name='None', txpower=None, uuid16s=[], uuid128s=[], rssi=-50, mfg_data=b'\x99\x04\x05\x10vJ;\xc3\xa0\xff\xc4\xff\xfc\x03\xfc]\x96<t\x14\xc7\xdaE\xd2E\xbb')
2023-11-30 20:38:11,770 DEBUG -                bleson.py: 26 - _run_get_data_background(): Data: Advertisement(flags=0x06, name='Ruuvi 45BB', txpower=None, uuid16s=[], uuid128s=[UUID128('6e400001-b5a3-f393-e0a9-e50e24dcca9e')], rssi=-50, mfg_data=None)
````

So, the rssi seems to be there. When I have time, I'll see if I can find a way to extract that data.

sevesalm, 2023-12-02:
I created a PR for this in #235 .

closed 2023-12-06 by sevesalm