ttu/ruuvitag-sensor #237: Bluetooth device argument ignored when using Bleak

terop, 2024-02-12:
When using the Bleak communication module the Bluetooth device argument value is ignored.
For example I modified the `examples/get_async_bleak.py` example as shown below to use a non-existent device
and it still returned data which it shouldn't have done in case the argument value was used.
```
diff --git a/examples/get_async_bleak.py b/examples/get_async_bleak.py
index 40e65ff..b34ce51 100644
--- a/examples/get_async_bleak.py
+++ b/examples/get_async_bleak.py
@@ -10,7 +10,7 @@ ruuvitag_sensor.log.enable_console()
 
 
 async def main():
-    async for data in RuuviTagSensor.get_data_async():
+    async for data in RuuviTagSensor.get_data_async(bt_device="foo"):
         print(data)
```

The pertinent question is thus how to deal with this problem. IMO the minimum solution is to update the documentation to note that with Bleak the Bluetooth device argument has no effect. However should something also be done and in case yes, what?

**Environment:**
 - OS: OS independent problem, tested on Gentoo
 - ruuvitag_sensor package version: master
 - RuuviTag firmware version: latest
 - Data format: 5

ttu, 2024-02-14:
Hi! On Linux Bleak supports different adapters/devices.

In the current `BleCommunicationBleak` implementation `bt_device`-parameter is not passed to the `BleakScanner`
https://github.com/ttu/ruuvitag-sensor/blob/8be0d086aa96f7f6b75270291a50893775ce7f1e/ruuvitag_sensor/adapters/bleak_ble.py#L62C51-L62C60

https://github.com/ttu/ruuvitag-sensor/blob/8be0d086aa96f7f6b75270291a50893775ce7f1e/ruuvitag_sensor/adapters/bleak_ble.py#L29

`BleakScanner` with Linux has a support for `adapter`-argument
https://github.com/hbldh/bleak/blob/f1b8425b7e2457cf1020ae83860e88905123d0f2/bleak/backends/scanner.py#L274
https://github.com/hbldh/bleak/blob/f1b8425b7e2457cf1020ae83860e88905123d0f2/bleak/backends/bluezdbus/scanner.py#L128

Would you like to make the change to `BleCommunicationBleak` and pass `bt_device` to `BleakScanner` if it is defined?

terop, 2024-02-14:
PR opened: https://github.com/ttu/ruuvitag-sensor/pull/238.
There is some problem in the CI which I cannot solve myself.

terop, 2024-02-27:
As PR https://github.com/ttu/ruuvitag-sensor/pull/238 is now merged this issue is now done.

closed 2024-02-27 by terop