ttu/ruuvitag-sensor #241: Consider using Ruff to replace flake8, isort, PyLint and Black

terop, 2024-02-17:
[Ruff](https://docs.astral.sh/ruff/) is a "multi-purpose" tool for Python linting and code-formatting which can replace the current tools apart from mypy. This would improve the developer experience as there would be only two tools for linting and formatting instead of four that there is now. Ruff is also very fast which makes it pleasant to use.

There are some downsides too: Ruff does not yet have all the rules which flake8 and PyLint  includes. Also it is still under somewhat active development meaning that previously working code might be "broken" with a newer Ruff version. This problem can of course be avoided by using a pinned version which probably should be done anyway.

Would this idea make sense?

terop, 2024-02-17:
This is not a real bug but an enhancement / idea but I couldn't figure out how to change the label to something else than a bug. Please feel free to change the label to something more appropriate.

ttu, 2024-02-19:
It seems that Ruff is already quite production-ready and is used by many large projects, so I agree that we should give it a try.

Since it is under very active development, we should indeed pin the version we use, as you suggested.

Would you like to try it and share your findings here?

terop, 2024-02-19:
Sure, at some point.

PR #258 (2024-12-28): chore: use ruff for linting and code formatting
ruff replaces black, flake8, isort and pylint.

Fixes #241.
closed 2024-12-28 by ttu