ttu/ruuvitag-sensor #245: Release version 3.0

ttu, 2024-02-19:
- [ ] Bleak as default adapter for Linux
  - https://github.com/ttu/ruuvitag-sensor/pull/244
- [ ] Drop support for Python 3.7

terop, 2024-02-19:
I created a commit for item two in the above list,  it can be found at https://github.com/terop/ruuvitag-sensor/commit/757b441cb7f6e7b0edbc48e87552ac8b100a2ae2.
@ttu please take a look in case I missed something and if something should be change or improved.

ttu, 2024-02-24:
Thanks, commit looks good. I will leave comments if something comes to mind that is missing.

I was thinking that we should also drop support for 3.8 with this release. Its end of life is in October 2024, so by the time of the release, it won't be that far away. Based on the PyPI stats, it is not very widely used.

terop, 2024-03-11:
@ttu Could you kindly check the above commits?

ttu, 2024-04-04:
@terop sorry again for the delay. I think those commits already had pretty much all required changes. I found few mentions about default adapters that should be changed to Bleak, but after that I think those are good 👍

ttu, 2024-07-19:
@terop sorry for the delay again. I finally had some time to get back to this. 

I created two PRs; selecting default adapter and updated verification script. Could you create a PR for dropping Python 3.7 & 3.8 support?

I will still go through the functionality and test this with Linux to make sure we are not missing anything.

terop, 2024-07-23:
@ttu Sure and sorry for the delay.

Please find a PR to drop Python 3.7 and 3.8 at https://github.com/ttu/ruuvitag-sensor/pull/251.

terop, 2024-11-09:
Now all currently known tasks are done.

ttu, 2024-11-23:
@terop I was performing verifications before the release. I successfully ran `verification.sh` on macOS and Windows, but my Linux machine is experiencing some issues with Bluetooth. Could you run the verification from the master branch using Bleak and BlueZ on Linux?

terop, 2024-11-24:
@ttu Running `verification.sh` using BlueZ works fine but I get some error with Bleak. I will quickly check if I can figure what causes it.

terop, 2024-11-24:
I did some more digging on this problem and it seems that https://github.com/ttu/ruuvitag-sensor/blob/master/ruuvitag_sensor/adapters/bleak_ble.py#L104 is possibly not called at the return from a Bleak async function. From the output of `verification_async.py` (see below) one can see that `scanner.start()` is called but the log message for the stop is not printed. Of course it is possible that the `stop()` is called but the log message is not printed. However my knowledge of Python async functions and Bleak API is insufficient to say what actually happens.

```
...
############################################
RuuviTagSensor.get_data_for_sensors_async
############################################
Get latest data for sensors. Stop with Ctrl+C.
Stops automatically in 5s
MACs: []
Bleak scanner started
<prints removed>
{'FB:15:98:DF:A0:A3': {'data_format': 5, ...}
OK
...
```

ttu, 2024-11-26:
@terop thanks for the info. With that it was pretty easy to debug the problem.

I noticed that `GeneratorExit` was not raised immediately when the async for loop was exited, which resulted in `scanner.stop` not being called immediately after the exit.

I made a change to explicitly close the generator after breaking the async for loop.

Can you try to execute verifications from this branch?
https://github.com/ttu/ruuvitag-sensor/tree/fix-close-generator-after-async-loop

terop, 2024-11-26:
@ttu Still broken with the same error. However I did notice the the branch you mentioned  contains the same commits as master branch so I assume something is missing.

terop, 2024-11-26:
@ttu Now it works, thank you!

Could you commit the fix soon? I would like to use it to get rid of the hack described above.

terop, 2024-11-27:
@ttu Unfortunately your commit from yesterday didn't fix the problem as I had hoped so I am asking in case you have any ideas how to fix this.

I have a async function which runs `RuuviTagSensor.get_data_async()` with a given set of RuuviTag MACs with `asyncio.wait_for()` having a timeout. In case one or more of the tags is not found the `wait_for()` call will fail with `TimeoutError` and the Bleak scanner is not stopped. Thus is there a way to stop the Bleak scanner in this situation so that `get_data_async()` could be called again?

ttu, 2024-11-28:
@terop Can you provide me an example code how you are calling `get_data_ascync` multiple times.

I suppose Linux has some limitations on using multiple instances of `BleakScanner`. I suppose this code doesn't work on Linux?
```py
import asyncio
from typing import AsyncGenerator

import ruuvitag_sensor.log
from ruuvitag_sensor.ruuvi import RuuviTagSensor
from ruuvitag_sensor.ruuvi_types import MacAndSensorData

ruuvitag_sensor.log.enable_console()

async def process_data_from_iter(data_iter: AsyncGenerator[MacAndSensorData, None], source_name: str):
    async for data in data_iter:
        print(f"Data from {source_name}: {data[0]}")

async def main():
    data_iter1 = RuuviTagSensor.get_data_async()
    data_iter2 = RuuviTagSensor.get_data_async()

    # Process both iterators concurrently
    task1 = asyncio.create_task(process_data_from_iter(data_iter1, "Iterator 1"))
    task2 = asyncio.create_task(process_data_from_iter(data_iter2, "Iterator 2"))

    await asyncio.gather(task1, task2)

if __name__ == "__main__":
    asyncio.run(main())
```

Maybe for Linux I have to try to find a way to force close previous scanner if `Bleak.get_data` is called multiple times.

terop, 2024-11-29:
I don't think it is worth checking this problem any further as most of the time the tag is not found during a second scan either.

ttu, 2025-01-13:
Released: https://github.com/ttu/ruuvitag-sensor/releases/tag/3.0.0

closed 2025-01-13 by ttu